johannes-graeter/limo #1: no `matches_msg_types`

extr15, 2018-04-24:
Hi @johannes-graeter , Thanks for your project.
I'd like to have a try, but when compiling, it seems I don't have `matches_msg_types`, is this an internal project?
could you help me? 
Thanks again!

johannes-graeter, 2018-04-24:
Hi @extr15,
thank you for your interest in my work.
I will look into it and try to provide a working example until tomorrow.

Regards,

Johannes

johannes-graeter, 2018-04-24:
Hi there,
I included the missing headers.
For a small example have a look at the unittests.
Tomorrow I will write better documentation and test the dependencies on a clean machine.
Please note that the depth estimation needed for the Kitti results is not published yet, but will be soon.

closed 2018-04-24 by johannes-graeter

extr15, 2018-04-25:
@johannes-graeter , Thanks for your reply.
I encounter an error:
```
landmark_selection_scheme_voxel.hpp:28:59: error: array must be initialized with a brace-enclosed initializer
         std::array<double, 3> voxel_size_xyz{1.0, 1.0, 0.5}; ///< Voxelsize in x,y,z in meters
```
and I fixed it by `double-braces`:
```
        std::array<double, 3> voxel_size_xyz { {1.0, 1.0, 0.5} }; ///< Voxelsize in x,y,z in meters
```
now it compile OK.
Looking forward to your progress!

johannes-graeter, 2018-04-25:
Hi,
thanks for the fix. This is a matter of compilers. My gcc c++14 uses brace elision and can cope with that syntax (see https://stackoverflow.com/questions/16985687/brace-elision-in-stdarray-initialization). 
Out of curiosity what compiler  and c++ version are you using?

extr15, 2018-04-26:
Hi, thanks for the link. I use `gcc 5.4.0`, it seems I don't specify the c++ version.

johannes-graeter, 2018-04-26:
seems the default fpr 5.4.0 is c++11 https://gcc.gnu.org/gcc-5/changes.html .
If you can, consider changing to gcc 7, it is really worth it. Compiler messages are way better understandable and the warnings are way more intelligent and usefull!

extr15, 2018-04-26:
Thanks for your suggestion! I may try later :smiley:

johannes-graeter, 2018-05-22:
the complete core library and most of the packages necessary for pre processing. Since I tried to make everything modular its quite a lot of packages. I would be happy if you share your experience following the install instructions and the reults of the algorithm.