johannes-graeter/limo #5: CMake Error :  Package mrt_ceres

mrgransky, 2018-07-25:
Hi,

I followed your instruction to build the project, I get the following error after `catkin build` command in `install_repos.sh` file:

```
Errors     << monolidar_fusion:cmake /home/farid/catkin_ws/logs/monolidar_fusion/build.cmake.004.log                                    
CMake Warning at /home/farid/catkin_ws/devel/share/mrt_cmake_modules/cmake/mrt_cmake_modules-extras.cmake:10 (message):
  Current CMakeLists.txt version is less than the recommended version 2.1.
  If you are the maintainer, please update it with

  'mrt maintenance update_cmakelists monolidar_fusion'.
Call Stack (most recent call first):
  /home/farid/catkin_ws/devel/share/mrt_cmake_modules/cmake/mrt_cmake_modulesConfig.cmake:197 (include)
  CMakeLists.txt:9 (find_package)

CMake Error at /home/farid/catkin_ws/src/mrt_cmake_modules/cmake/Modules/FindAutoDeps.cmake:93 (message):
  Package mrt_ceres: Specified include dir variable CERES_INCLUDE_DIRS not
  set.
Call Stack (most recent call first):
  CMakeLists.txt:16 (find_package)

cd /home/farid/catkin_ws/build/monolidar_fusion; catkin build --get-env monolidar_fusion | catkin env -si  /usr/bin/cmake /home/farid/catkin_ws/src/mono_lidar_depth/monolidar_fusion --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/farid/catkin_ws/devel/.private/monolidar_fusion -DCMAKE_INSTALL_PREFIX=/home/farid/catkin_ws/install; cd -
........................................................................................................................................
```

I have also installed ceres from the link provided.

Do you have any idea what I am really missing in here?

Cheers mate;
Farid

johannes-graeter, 2018-07-25:
Hey,

thanks for the interest in my work.
At the moment I am busy writing my thesis, so maybe I will not respond very quickly ;)

Hm... mrt_ceres should link to the local install...
Try changing "mrt_ceres" in the package.xml to "ceres", does it work now?

Regards

mrgransky, 2018-07-25:
Hi,

In monolidar_fusion package, I applied the following modification as you mentioned for package.xml:

`<depend>mrt_ceres</depend>` to ` <depend>ceres</depend>`. The error looks like this now:

```
Errors     << monolidar_fusion:cmake /home/farid/catkin_ws/logs/monolidar_fusion/build.cmake.000.log                                    
CMake Warning at /home/farid/catkin_ws/devel/share/mrt_cmake_modules/cmake/mrt_cmake_modules-extras.cmake:10 (message):
  Current CMakeLists.txt version is less than the recommended version 2.1.
  If you are the maintainer, please update it with

  'mrt maintenance update_cmakelists monolidar_fusion'.
Call Stack (most recent call first):
  /home/farid/catkin_ws/devel/share/mrt_cmake_modules/cmake/mrt_cmake_modulesConfig.cmake:197 (include)
  CMakeLists.txt:9 (find_package)

CMake Error at /home/farid/catkin_ws/src/mrt_cmake_modules/cmake/Modules/FindAutoDeps.cmake:80 (message):
  Package ceres is specified for autodepend but cmake variables are not
  defined.
Call Stack (most recent call first):
  CMakeLists.txt:16 (find_package)
```

johannes-graeter, 2018-07-30:
Hey,

so I tried to reproduce your error, but I didn't succeed.
I set up a new virtual box and installed everything from scratch and it did get the dependency mrt_ceres.
How did you install ceres? Did you use the linked instructions of ceres in my README.md?
For your environment to find it you need to "sudo make install" the ceres headers as described in the instructions, did you do that?

Best regards,

Johannes

mrgransky, 2018-07-31:
Hi,

This is the procedures I followed to install ceres solver from the link provided in README.md:

First, I installed all dependencies as follows (some of which were already installed in my machine): 

```
# CMake
$ sudo apt-get install cmake
# google-glog + gflags
$ sudo apt-get install libgoogle-glog-dev
# BLAS & LAPACK
$ sudo apt-get install libatlas-base-dev
# Eigen3
$ sudo apt-get install libeigen3-dev
# SuiteSparse and CXSparse (optional)
# - If you want to build Ceres as a *static* library (the default)
#   you can use the SuiteSparse package in the main Ubuntu package
#   repository:
$ sudo apt-get install libsuitesparse-dev
# - However, if you want to build Ceres as a *shared* library, you must
#   add the following PPA:
$ sudo add-apt-repository ppa:bzindovic/suitesparse-bugfix-1319687
$ sudo apt-get update
$ sudo apt-get install libsuitesparse-dev
```
Second, I followed the instructions to install `ceres-solver`:
```
$ git clone https://ceres-solver.googlesource.com/ceres-solver
$ cd ceres-solver
$ mkdir build
$ cmake ..
$ make
$ sudo make install
```

now I have a package, called `ceres` located at `/usr/local/include`. The screenshot of my environment validates that I have installed `ceres`. But I do not have any package called `mrt-ceres `as you have explained in README.md.

[image]

[image]

I am also able to run the sample example with ceres as explained in [here](http://ceres-solver.org/installation.html).

[image]

The error I get is related to two packages, `monolidar_fusion` and `keyframe_bundle_adjustment` as you can see in here:

[image]

As I explored the `package.xml` in those packages, I have realized there is `mrt-ceres` as dependecy which returns the error if I am not entirely mistaken.

The screenshots of those files are as follows:

[image]

and 
[image]

Your help would be genuinely appreciated... :)

johannes-graeter, 2018-07-31:
Ok lets get to it:
in install_repos.sh all repos used are downloaded from github.
the one responsible for the package deps is mrt_cmake_modules .
In that repo a find script is defined for mrt_ceres, which looks for a custom build and if you do not have that (which should be the case for you) it uses the "normal" find script from ceres:

https://github.com/KIT-MRT/mrt_cmake_modules/blob/master/cmake/Modules/FindMrtCeres.cmake

so cmake should say "Warning: Mrt-Ceres package not found. Using System version instead." and link to the one you installed manually.
The find_package command looks then for a FindCeres.cmake config in the default path and sets CERES_INCLUDE_DIRS.
That is used for the base.yaml in the mrt_cmake_modules (https://github.com/KIT-MRT/mrt_cmake_modules/blob/master/yaml/base.yaml) to add the ceres include dirs to the compiler.
I am not sure why cmake apparently can not find the FindCeres.cmake...
I guess somehow your cmake paths seem not to be set correctly...

As a quick hack you could manually add the path to the base yaml, just subsitute CERES_INCLUDE_DIRS by /usr/local/include/ceres and the library path accordingly.

johannes-graeter, 2018-07-31:
You can also have a look at the virtual machine (make sure to set the keyboard to english ;) ) on which the program compiles.
It has ceres installed in /usr/include but that souhldn'T be much of a difference.

johannes-graeter, 2018-07-31:
btw I used the latest stable release, so

tar zxf ceres-solver-1.14.0.tar.gz
mkdir ceres-bin
cd ceres-bin
cmake ../ceres-solver-1.14.0
make -j3
make test
# Optionally install Ceres, it can also be exported using CMake which
# allows Ceres to be used without requiring installation, see the documentation
# for the EXPORT_BUILD_DIR option for more information.
sudo make install

but unless ceres github repo does sth strange with its cmake find scripts that shouldn'T be the problem...

poggenhans, 2018-08-01:
Hey @mrgransky,

the issue was related to a [recent change in ceres](https://github.com/ceres-solver/ceres-solver/commit/2166bad133d279ec28008f1df8b4d90a7023f531#diff-78a25a1f160d557a7a40b1a9da11a15c).

We fixed that now in the mrt_cmake_modules package. Could you update that and test if it works now?