johannes-graeter/limo #6: Code for epipolar error

rising-turtle, 2018-08-21:
Hi All:

In the paper for equation (3) where the epipolar error is computed to add constraints between two features without depth. But I cannot find where the code is that for the epipolar error computation. Maybe I just missed it, or can anyone shows me where the code is ? 

Thanks !

johannes-graeter, 2018-08-21:
Nice catch, yeah you are right!
In the paper I used a separate frame to frame motion estimation algorithm (Liviodo), which used the epipolar constraint. Since that module would have added 5 more libraries to the project I decided to stay lean and not add it.
In the version as is, I use 5 point-algo for rotation and the old map from the BA to do a motion only adjustment as frame to frame motion estimation which results in a little less quality of the algorithm but saves around 70 ms delay.
LIMO2 as published on KITTI uses only that and has even slightly better results. 
As soon as I am back from vacation I will update the repositories for LIMO2.

Thanks for the interest in my project!

Regards,

Johannes

rising-turtle, 2018-08-21:
@johannes-graeter 
Thanks for quick reply. Look forward to your new update. 
I have a question, since fundamental matrix F= KEK', with essential matrix E = R[t]x, meaning that both rotation [R] and translation [t] are incorporated to compute F. However, it seems that in your paper  F(x/z, y/z, r, p, y) uses x/z, y/z rather x,y,z? And in this way, F is related to translation component t= [x,y,z], and therefore the epipolar error will constraint not only rotation but also translation? Please correct me if my understanding is not right. 

Thanks in advance. 

David

johannes-graeter, 2018-08-21:
Hi there,
yeah you understand that correctly. The epipolar geometry constraints 5 degrees of freedom (angles and 2 dof of the translation vector). Scale (the length of the translation vector) can not be constrained for one single camera. 

So yes, the direction of the translation can be calculated but not its length.
Whereas the epipolar constraint works well for rotation, it is rather ill conditioned for translation, which is why state of the art odometry algorithms use either bundle adjustment or use only the rotation estimation from the epipolar constraint (f.e. RotRocc++ on KITTI).

In the past I did some work on that: 
https://github.com/johannes-graeter/momo
Where I played with some variants of the epipolar error for multi camera motion estimation.
There is also code for the epipolar error for pinhole cameras and fisheye cameras :)

Regards,

Johannes

PS: Szeliski: Computer Vision is great http://szeliski.org/Book/

rising-turtle, 2018-08-21:
@johannes-graeter 

That's of great help. Thanks for your reply. 

Best, 

David

closed 2018-08-21 by rising-turtle