johannes-graeter/limo #7: get error of tracklet_depth_node-3

aicarmark, 2018-09-25:
when start running, got error as below:

Got the camera info:
img_width: 1226
img_height: 370
focal_length_x: 707.091
principalPointX: 601.887
principalPointY: 183.11

Initialize DepthEstimator Post
DepthEstimator successful initialized
===============================================================================**=REQUIRED process [tracklet_depth_node-3] has died!
process has died [pid 23803, exit code -11, cmd /home/carmark/SLAM/limo-slam/devel_limo_release/lib/tracklets_depth_ros_tool/tracklet_depth_node __name:=tracklet_depth_node __log:=/home/carmark/.ros/log/ee4ed292-c0d5-11e8-848d-9cb6d0f8f3e7/tracklet_depth_node-3.log].
log file: /home/carmark/.ros/log/ee4ed292-c0d5-11e8-848d-9cb6d0f8f3e7/tracklet_depth_node-3*.log
Initiating shutdown!**
================================================================================
[mono_lidar-17] killing on exit
[semantic_labels-16] killing on exit
[label_dilation-15] killing on exit
[add_outlier_flag-14] killing on exit
[transform_local_cs_camera_gt-13] killing on exit
[transform_vehicle_camera_vehicle_gt-12] killing on exit
[transform_local_cs-11] killing on exit
[transform_local_estimate_prior-10] killing on exit
[transform_local_cs_left_camera_estimate_prior-9] killing on exit
[transform_vehicle_left_camera_estimate_prior-8] killing on exit
[ INFO] [1537888640.489776084, 0.528613523]: Duration feature matching and tracking: 71 ms
[transform_local_cs_vehcile_estimate-7] killing on exit
[transform_vehicle_left_camera_estimate-6] killing on exit
[viso_feature_tracking_node_left-2] killing on exit
[tf2_static_alias_vehicle_to_rig_left_estimate-4] killing on exit
[tf2_static_alias_vehicle_to_rig_right_estimate-5] killing on exit
[tracklet_depth_node-3] killing on exit
[gamma_correction_node_left-1] killing on exit
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
--------------------------------------
Dumped map to /tmp/mono_lidar_map_dump.yaml--------------------------------------

shutting down processing monitor...
... shutting down processing monitor complete
done

johannes-graeter, 2018-09-25:
Hi there,

have you tried running it in the vm is supplied?

aicarmark, 2018-09-26:
@johannes-graeter 
I checkout 
[d40600518ef6585b47f68773af5e72955e7dcdc0](https://github.com/johannes-graeter/limo/commit/d40600518ef6585b47f68773af5e72955e7dcdc0)
[4b3f6f38792c50417ae501ba00c33db87704a706](https://github.com/johannes-graeter/mono_lidar_depth/commit/4b3f6f38792c50417ae501ba00c33db87704a706)

got same erros

Initialize DepthEstimator Post
DepthEstimator successful initialized
[ INFO] [1537942026.394675088, 0.524007758]: Duration feature matching and tracking: 34 ms
================================================================================REQUIRED process [tracklet_depth_node-3] has died!
process has died [pid 19060, exit code -11, cmd /home/carmark/SLAM/limo-slam/devel_limo_release/lib/tracklets_depth_ros_tool/tracklet_depth_node __name:=tracklet_depth_node __log:=/home/carmark/.ros/log/a3422cbc-c14e-11e8-a97e-9cb6d0f8f3e7/tracklet_depth_node-3.log].
log file: /home/carmark/.ros/log/a3422cbc-c14e-11e8-a97e-9cb6d0f8f3e7/tracklet_depth_node-3*.log
Initiating shutdown!
================================================================================
[semantic_labels-16] killing on exit
[mono_lidar-17] killing on exit
[label_dilation-15] killing on exit
[add_outlier_flag-14] killing on exit
[transform_local_cs_camera_gt-13] killing on exit
[transform_vehicle_camera_vehicle_gt-12] killing on exit
[transform_local_cs-11] killing on exit
[transform_local_estimate_prior-10] killing on exit
[transform_vehicle_left_camera_estimate_prior-8] killing on exit
[transform_local_cs_left_camera_estimate_prior-9] killing on exit
[transform_local_cs_vehcile_estimate-7] killing on exit
[transform_vehicle_left_camera_estimate-6] killing on exit
--------------------------------------
Dumped map to /tmp/mono_lidar_map_dump.yaml--------------------------------------

johannes-graeter, 2018-09-26:
please turn verbosity to debug as described in the last comment. 
Like that I can not see at what point the method fails.

Also try the virtual machine please.

johannes-graeter, 2018-09-27:
I forgot to add the nodelet_plugins file in that prjoect (see https://github.com/johannes-graeter/limo/issues/8)
pull mono_lidar_depth and try again please :)

closed 2018-11-09 by johannes-graeter