johannes-graeter/limo #8: Running demo, using 04.bag, I see nothing.

MontyTHall, 2018-09-26:
When I launch the limo node, I see the following flash by quickly
```[ERROR] [1537984722.112431485]: Skipped loading plugin with error: XML Document '/home/xzcjml/catkin_limo/src/mono_lidar_depth/tracklets_depth_ros_tool/nodelet_plugins.xml' has no Root Element. This likely means the XML is malformed or missing..```

and also 
```[ERROR] [1537984722.275178784]: "sensor/camera" passed to lookupTransform argument target_frame does not exist. ```

johannes-graeter, 2018-09-27:
I forgot to add the nodelet_plugins.xml -.- 
Should be fixed by now...

MontyTHall, 2018-09-27:
The error message went away but nothing shows up in rviz.  I just noticed that there's a virtual box image, I'm going to give that a spin.

zxhou, 2018-10-27:
I meet the same problem. The program is running but there is nothing shows up in rviz. Although I changed several fixed frame, it still doesn't work.

tuandle, 2018-10-29:
@johannes-graeter Thanks for open-source this. 
Here is what I got from Rviz. However, I am not sure what to make out of it. The trajectories (blue-active path and green-complete path) and estimate landmarks just stopped being outputted midway. Is that because I am running in debug mode? 
Also, is it possible that you share a video screen capture of your PC running the package with 04.bag? So that we can have a general idea of what to expect :)
[image]
Thanks again!

johannes-graeter, 2018-10-29:
By "debug mode" you mean that you compile the program with debug compiler flags or do you compile it with the script I supply in the repo? "debug" ros output to the terminal should not be a problem.
What kind of machine are you running it on? (CPU specs and RAM)

tuandle, 2018-10-29:
@johannes-graeter yes, I used your scripts. 
My laptop specs is like this:
```
tuan@tuan:~$ sudo lshw -short
H/W path         Device     Class          Description
======================================================
                            system         GL502VSK
/0                          bus            GL502VSK
/0/0                        memory         64KiB BIOS
/0/8                        memory         32GiB System Memory
/0/8/0                      memory         16GiB SODIMM Synchronous 2400 MHz (0,4 ns)
/0/8/1                      memory         [empty]
/0/8/2                      memory         16GiB SODIMM Synchronous 2400 MHz (0,4 ns)
/0/8/3                      memory         [empty]
/0/e                        memory         256KiB L1 cache
/0/f                        memory         1MiB L2 cache
/0/10                       memory         6MiB L3 cache
/0/11                       processor      Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz
```

johannes-graeter, 2018-10-29:
Ok thats odd. Any error message thrown when the trajectory stops being published?

johannes-graeter, 2018-10-29:
thanks for the screencast, that was helpful.
so the problem is that even though the specs of your computer look ok, the optimization for the full bundle adjustment each time a  keyframe is chosen takes way too long (around 1 sec which is 4 times more than on my computer with similar specs). That's why after a while queues of the rosnodes overflow and data gets lost... This is a bit hard to debug from remote though... First thing you can check is if ceres is build in release mode. Did you build it on your machine or did you use the prebuilts from apt-get install? We'll get to it ;)

johannes-graeter, 2018-10-29:
made a new topic since this got off-topic.
@tuandle please respond there.

closed 2018-11-09 by johannes-graeter

gychen-n, 2021-12-29:
> The error message went away but nothing shows up in rviz. I just noticed that there's a virtual box image, I'm going to give that a spin.

Hello, have you solved this error? How?
`[ERROR] [1537984722.275178784]: "sensor/camera" passed to lookupTransform argument target_frame does not exist.`