johannes-graeter/limo #10: Minimum Topics to get LIMO to run and what launch file to use?

MontyTHall, 2018-10-17:
LIMO seems to be quite large and I'd like to get it to work with my LIDAR and Camera as soon as possible.  So what I did was playback 04.bag to see what topics are to be published and mimic that.

For my Camera:
1. I created 2 image, one black and white (image 1) and the other color (image 2).  04 bag seems to be stereo so what I did was publish left and right color with the same image (image 2), and left and right that's lack and white (image 1). topics: /sensor/camera/(color|grayscale)/(left|right)/image_rect, frame_ids:/sensor/camera/(color|grayscale)/(left|right)
2. I published 4 camera info: topics: /sensor/camera/(color|grayscale)/(left/right)/camera_info, frame_ids:/sensor/camera/(color|grayscale)/(left|right)
3. I published 1 point cloud: topics: /sensor/velodyne/cloud_euclidean, frame_id = /sensor/velodyne

Is there anything else that I should do?  Do I need a robot state publisher and a URDF too?  If so, what would the xml look like?

Thanks

johannes-graeter, 2018-10-17:
Hi there,
thanks for the interest in my work :)

1. Stereo is not used, only left images are used for estimation, depending on your neural network for semantic segmentation either color or grayscale
2. same here
3. good

Things to do: 
* publish the transform from LIDAR to camera, f.e. with static_transform_publisher 
* you need to do semantic segmentation of the image with your favorite neural network (there is plenty of pretrained ones out there.)

for launchfiles please refrain to the demo_keyframe_bundle_adjustment package https://github.com/johannes-graeter/limo/tree/master/demo_keyframe_bundle_adjustment_meta/launch .
Everything that is launch can be found in there.

What kind of LIDAR are you using? Depending on its density you might need to tune the depth extraction paramters.

Please tell me if you have questions or if it worked :) (pretty exciting to try it on a different system).

Best,

Johannes

johannes-graeter, 2018-10-18:
These are 2 possible projexts to use 
* https://github.com/tensorflow/models/tree/master/research/deeplab (easy to use since integrated in tensorflow, but I don't know how the runtime is.)
* https://github.com/itijyou/ademxapp (Closest to the one we are using)
* I just talked with our Deep learning expert and we will probably publish the net I used in a few weeks.
* I will try to make a version without semantics, should work for LIDAR and mono, but not for Mono only.

I always put parameters in the launch files not in the yaml parameter files (since you need to tune them mostly anyways ;) ). You find the few parameters used here https://github.com/johannes-graeter/limo/blob/master/demo_keyframe_bundle_adjustment_meta/launch/semantic_labels.launch .
Be aware that this node only assigns the semantic labels to the tracklets, it doesn't do the semantic labelling itself (labelled images are part of the rosbag...)

johannes-graeter, 2018-10-18:
I just had a look at the AEye sensor... Super interesting! However, since it seems very different to the Velodyne HDL 64 I used, some tweeking and tuning will be necessary. 
Since I am super thrilled by the possible application to that sensor, I would propose to try to make it run on your data if you are allowed to give the data to me. If that is ok for you just send me a rosbag with your data (camera image, 3d pointcloud, intrinsic calibration, extrinsic calibration) and I will make it run :)

johannes-graeter, 2018-11-09:
Any advancements? I would really like to know how my stuff works with AEye :) Thanks!

closed 2019-01-09 by johannes-graeter