johannes-graeter/limo #12: optimization is taking too long

johannes-graeter, 2018-10-29:
@johannes-graeter No, I don't think there is any error output, except for the very beginning when it complains about `sensor/camera` frame can't be found. I capture my screen [here](https://youtu.be/xO6ik85WNyI).

The final screenshot is like this:
[image]
So is this the expected output?

_Originally posted by @tuandle in https://github.com/johannes-graeter/limo/issues/8#issuecomment-433929329_

tuandle, 2018-10-29:
@johannes-graeter thanks for your fast reply. I built ceres from source quite some time ago and I can't remember if I built it in release mode. Nonetheless, I uninstalled and rebuilt ceres with `-DCMAKE_BUILD_TYPE=Release` flag (I am not sure if it has any effect since the current installation guide doesn't mention it). I also `catkin clean` the workspace and rebuild everything but the problem still persists.

tuandle, 2018-10-29:
@johannes-graeter I decided to switch to a desktop computer and I have converged solutions from ceres. I am not sure what went wrong with my laptop but the problem definitely not from the package. 
The final result looks fine! Thanks again! I guess I have to do a fresh install on my laptop.
[image]

johannes-graeter, 2018-10-29:
Cool! Thanks for making it work!
If you know the issue please share :) 
If you have some experience with the system I would be happy to know what works, what doesn't (people tend to give feedback on setup issues, but I haven't had feedback on a working setup yet ;) )

KleinYuan, 2021-11-26:
I have met the same issue:
- at the first place, it worked
- after I installed other packages, which potentially touches glog and gflags, which are all important dependencies for ceres, I observed the issue happened
- then, I nuked the workspace and reinstalled the ceres&limo, the issue is gone

johannes-graeter, 2021-11-26:
@KleinYuan Thanks for trying out my stuff :) 
IN order to get a hold of the issue, could you try out the docker container and install the packages that conflicted in that clean environment, so we can make it repdoucible? Otherwise debugging will be quite hard

KleinYuan, 2021-11-27:
Unfortunately, the docker seems not working for me. The feature tracking repo has some broken changes, which pbbly shall be fixed for the docked image first.

I can try the docker later after I wrap up my current on-going project (on a ddl), to verify the RC.

johannes-graeter, 2022-07-31:
seems to work for others, closing until issue pops up again

closed 2022-07-31 by johannes-graeter