johannes-graeter/limo #15: /usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(node_data.cpp.o)

Chrislzy1993, 2019-01-07:
hi, when i run `bash install_repos.sh`, teminator display the following error message,  it is about `yaml-cpp`, but i dont know how to fix it
ERROR MESSAGE: 
```
Errors     << viso_feature_tracking_ros_tool:make /home/liuzhiyang/limo/logs_limo_release/viso_feature_tracking_ros_tool/build.make.001.log
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(node_data.cpp.o): requires dynamic R_X86_64_32 reloc which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(node_data.cpp.o): requires dynamic R_X86_64_32 reloc which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(node_data.cpp.o): requires dynamic R_X86_64_32 reloc which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(node_data.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_Znwm' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(node_data.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_Znwm' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(node_data.cpp.o): requires dynamic R_X86_64_32 reloc against '__dso_handle' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(node_data.cpp.o): requires dynamic R_X86_64_32 reloc against '__gxx_personality_v0' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(exceptions.cpp.o): requires unsupported dynamic reloc 11; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(memory.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_Znwm' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(memory.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_ZdlPv' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(memory.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_ZdlPv' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(memory.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_ZdlPv' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(memory.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_ZdlPv' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(memory.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_ZdlPv' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(memory.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_ZdlPv' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(memory.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_ZdlPv' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(memory.cpp.o): requires dynamic R_X86_64_32 reloc against '__pthread_key_create' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(memory.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_ZNSt8_Rb_treeIPN4YAML6detail4nodeES3_St9_IdentityIS3_ESt4lessIS3_ESaIS3_EE8_M_eraseEPSt13_Rb_tree_nodeIS3_E' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(memory.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_ZNSt8_Rb_treeIPN4YAML6detail4nodeES3_St9_IdentityIS3_ESt4lessIS3_ESaIS3_EE8_M_eraseEPSt13_Rb_tree_nodeIS3_E' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(memory.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_ZSt18_Rb_tree_decrementPSt18_Rb_tree_node_base' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(memory.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_ZSt18_Rb_tree_decrementPSt18_Rb_tree_node_base' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(memory.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_ZNSt8_Rb_treeISt10shared_ptrIN4YAML6detail4nodeEES4_St9_IdentityIS4_ESt4lessIS4_ESaIS4_EE29_M_get_insert_hint_unique_posESt23_Rb_tree_const_iteratorIS4_ERKS4_' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(memory.cpp.o): requires dynamic R_X86_64_32 reloc which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(parse.cpp.o): requires dynamic R_X86_64_32 reloc which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(parse.cpp.o): requires dynamic R_X86_64_32 reloc which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(parse.cpp.o): requires dynamic R_X86_64_32 reloc against '__pthread_key_create' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(parse.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_Znwm' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(parse.cpp.o): requires dynamic R_X86_64_32 reloc which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(parser.cpp.o): requires dynamic R_X86_64_32 reloc which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(parser.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_ZN4YAML9Exception10build_whatERKNS_4MarkERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(parser.cpp.o): requires dynamic R_X86_64_PC32 reloc against 'memcmp' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(parser.cpp.o): requires dynamic R_X86_64_PC32 reloc against 'memcmp' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(parser.cpp.o): requires dynamic R_X86_64_PC32 reloc against '_Znwm' which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(parser.cpp.o): requires dynamic R_X86_64_32 reloc which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(parser.cpp.o): requires dynamic R_X86_64_32 reloc which may overflow at runtime; recompile with -fPIC
/usr/bin/ld.gold: error: /usr/local/lib/libyaml-cpp.a(scanner.cpp.o): requires dynamic R_X86_64_32 reloc which may overflow at runtime; recompile with -fPIC
```

johannes-graeter, 2019-01-09:
Hi there,
seems as if your yaml-cpp is built against the wrong architecture.
Can you use yaml-cpp in other cpp projects?
First guess would be to make sure yaml-cpp is up to date with
sudo apt update
sudo apt upgrade

(you can also do it with synaptic if you want to only update yaml-cpp; I have version 0.5v5)

Regards,

Johannes

PS: I will not be able to answer frequently since my PhD defence will come up the 2.2. and I have to prepare...

Chrislzy1993, 2019-01-17:
thanks for your replay, i build and install yaml-cpp from github project.  i use 0.5.X version, i can use yaml-cpp sucessful in other project. thanks again

johannes-graeter, 2019-01-23:
That really seems to be an issue with yaml-cpp one solution I found is to download a recompiled version:
https://github.com/scylladb/scylla/issues/3221

Good luck :) Let me know if this works.

closed 2019-04-30 by johannes-graeter