johannes-graeter/limo #16: A question about the dynamic objects。

songjiaxina, 2019-01-20:
Thanks for your excellent work! I want to test limo on my own data. SInce Identifying the dynamic objects by deep learning is a little hard. So I want to know how much influence does the dynamic object have on the accuracy of the track? Have you ever tested it without semantic labels? Great thanks!!

johannes-graeter, 2019-01-21:
Hi there,

thanks for your interest in my work :)
Outliers in sense of feature tracks that have very high error to the solution are down-weighted using an M-Estimator. Moreover to make the optimization faster, I employ a trimmed-least-squares like optimization strategy, so after 4 iterations of the solver I erase the a percentage of measurements with high error (typically 5-10%). 

So if the dynamic objects are small enough (majority of the measurements is on the static scene), it should work. You can also play with the m-estimator parameter (https://github.com/johannes-graeter/limo/blob/master/demo_keyframe_bundle_adjustment_meta/launch/keyframe_ba_monolid.launch , line 45). Note that for depth and reprojection error the "robust_loss_threshold" must be tuned individually.

I tried that on KITTI and got a translation error around 1.05%. 

To do that you can just deactivate the "semantic_labels" node in kitti_standalone.launch (https://github.com/johannes-graeter/limo/tree/master/demo_keyframe_bundle_adjustment_meta/launch) and hand over a different tracklets_subscriber_topic to "keyframe_ba_monolid.launch".

If you tried it please share your result :)

Regards,

Johannes

songjiaxina, 2019-01-21:
Thanks for your detail answers! I will remember these valuable suggestions.Once I have tested limo on my own car, I will give you feedback in time.

johannes-graeter, 2019-01-25:
I there. Motivated by your issue I am testing limo at the moment on our own car without semantics.
Since the algorithm was developed on KITTI I ran into a few minor challenges, which however seem to be hard to solve if you are new to the software stack.
I will make it work on our car, fix the issues and make a small tutorial how to implement it with your own data. Probably I will need two weeks for it in order to be tested properly.

Cheers,
Johannes

johannes-graeter, 2019-02-06:
It runs nicely without the semantics :) please wait a little longer until the update of the repo :)

songjiaxina, 2019-02-09:
OK. Thanks! Wait for your update!

closed 2019-02-09 by songjiaxina

zoumaguanxin, 2019-04-28:
> It runs nicely without the semantics :) please wait a little longer until the update of the repo :)

waitting.

Hi,

I simply provide all pixels with the same label. Is it a solution without semantic labels?

johannes-graeter, 2019-04-28:
If you don't assign the labels they should be initialized to -1 and hence all treated equally. If you assign an arbitrary one, it risk to be one of the outlier labels.

zoumaguanxin, 2019-04-29:
There are two screen shots in the following that recorded the information when I deactivated the launch about semantic labels. The first one recorded the situation where limo is running in kitti dataset, the second is running in the demo bag provided by you. I found that some nodes seem not to work well. In ros, this situation happens usually when the required topic is not published correctly. I think that I must miss some important step, I really urge to run limo without semantic labels. It will be grateful to accept your any advice.  
[image]

[image]