johannes-graeter/limo #18: Build error in "bash install_repos.sh"

SRainGit, 2019-02-19:
Hi, I am trying to run your code. But when I followed the instruction to build repos `bash install_repos.sh` I got errors:

`Errors     << rosinterface_handler:make /media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/logs_limo_release/rosinterface_handler/build.make.000.log
Traceback (most recent call last):
  File "/media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/src/rosinterface_handler/test/cfg/DefaultsMissing.rosif", line 14, in <module>
    exit(gen.generate("rosinterface_handler", "rosinterface_handler_test", "DefaultsMissing"))
  File "/media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/src/rosinterface_handler/src/rosinterface_handler/interface_generator_catkin.py", line 562, in generate
    return self._generateImpl()
  File "/media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/src/rosinterface_handler/src/rosinterface_handler/interface_generator_catkin.py", line 571, in _generateImpl
    self._generatepy()
  File "/media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/src/rosinterface_handler/src/rosinterface_handler/interface_generator_catkin.py", line 947, in _generatepy
    with open(init_file, 'wa') as f:
ValueError: must have exactly one of create/read/write/append mode
Traceback (most recent call last):
  File "/media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/src/rosinterface_handler/test/cfg/Defaults.rosif", line 41, in <module>
    exit(gen.generate("rosinterface_handler", "rosinterface_handler_test", "Defaults"))
  File "/media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/src/rosinterface_handler/src/rosinterface_handler/interface_generator_catkin.py", line 562, in generate
    return self._generateImpl()
  File "/media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/src/rosinterface_handler/src/rosinterface_handler/interface_generator_catkin.py", line 571, in _generateImpl
    self._generatepy()
  File "/media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/src/rosinterface_handler/src/rosinterface_handler/interface_generator_catkin.py", line 947, in _generatepy
    with open(init_file, 'wa') as f:
ValueError: must have exactly one of create/read/write/append mode
make[2]: *** [/media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/devel_limo_release/.private/rosinterface_handler/include/rosinterface_handler/DefaultsMissingInterface.h] Error 1
make[2]: *** Deleting file '/media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/devel_limo_release/.private/rosinterface_handler/include/rosinterface_handler/DefaultsMissingInterface.h'
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [/media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/devel_limo_release/.private/rosinterface_handler/include/rosinterface_handler/DefaultsInterface.h] Error 1
make[2]: *** Deleting file '/media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/devel_limo_release/.private/rosinterface_handler/include/rosinterface_handler/DefaultsInterface.h'
......
Failed     << viso2:cmake                                         [ Exited with code 1 ]            
Failed    <<< viso2                                               [ 2.8 seconds ]                   
[build] Summary: 2 of 20 packages succeeded.                                                        
[build]   Ignored:   3 packages were skipped or are blacklisted.                                    
[build]   Warnings:  None.                                                                          
[build]   Abandoned: 10 packages were abandoned.                                                    
[build]   Failed:    8 packages failed.                                                             
[build] Runtime: 9.1 seconds total.                                                                 
[build] Note: Workspace packages have changed, please re-source setup files to use them.`

According to the error 'ValueError: must have exactly one of create/read/write/append mode', I changed the source code in 'interface_generator_catkin.py' by replacing ` with open(init_file, 'wa') as f:` with ` with open(init_file, 'w') as f:`. Then I got other errors:

`Errors     << rosinterface_handler:make /media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/logs_limo_release/rosinterface_handler/build.make.001.log                                                            
  File "/media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/devel_limo_release/.private/rosinterface_handler/share/rosinterface_handler/cfg/DefaultsAtLaunch.cfg", line 33
    <filter object at 0x7fa5be3dda20>.add(name = 'verbosity_param_wo_default', paramtype = 'str', level = 0, description = 'Sets the verbosity for this node', edit_method=verbosity_param_wo_default, default="0")
    ^
SyntaxError: invalid syntax
make[2]: *** [/media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/devel_limo_release/.private/rosinterface_handler/include/rosinterface_handler/DefaultsAtLaunchConfig.h] Error 1
make[2]: *** Waiting for unfinished jobs....
  File "/media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/devel_limo_release/.private/rosinterface_handler/share/rosinterface_handler/cfg/Defaults.cfg", line 26
    <filter object at 0x7f9c9e343c50>.add(name = 'int_param_w_default', paramtype = 'int', level = 0, description = 'An Integer parameter', edit_method="", default=1)
    ^
SyntaxError: invalid syntax
make[2]: *** [/media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/devel_limo_release/.private/rosinterface_handler/include/rosinterface_handler/DefaultsConfig.h] Error 1
make[1]: *** [CMakeFiles/rosinterface_handler_gencfg.dir/all] Error 2
make: *** [all] Error 2
cd /media/rain/Win10_E/SLAM/Codes/LIMO/catkin_ws/build_limo_release/rosinterface_handler; catkin build --get-env rosinterface_handler | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Failed     << rosinterface_handler:make                           [ Exited with code 2 ]                                                                                                                   
Failed    <<< rosinterface_handler                                [ 0.5 seconds ]                                                                                                                          
Abandoned <<< matches_msg_conversions_ros                         [ Unrelated job failed ]                                                                                                                 
Abandoned <<< keyframe_bundle_adjustment                          [ Unrelated job failed ]                                                                                                                 
Abandoned <<< image_preproc_ros_tool                              [ Unrelated job failed ]                                                                                                                 
Abandoned <<< matches_conversion_ros_tool                         [ Unrelated job failed ]                                                                                                                 
Abandoned <<< util_nodes_tf2_ros_tool                             [ Unrelated job failed ]                                                                                                                 
Abandoned <<< feature_tracking_core                               [ Unrelated job failed ]                                                                                                                 
Abandoned <<< tracklets_depth_ros_tool                            [ Unrelated job failed ]                                                                                                                 
Abandoned <<< viso_feature_tracking_ros_tool                      [ Unrelated job failed ]                                                                                                                 
Abandoned <<< keyframe_bundle_adjustment_ros_tool                 [ Unrelated job failed ]                                                                                                                 
Abandoned <<< demo_keyframe_bundle_adjustment_meta                [ Unrelated job failed ]
......
Failed     << image_preproc:cmake                                 [ Exited with code 1 ]                                                                                                                   
[... cut ...]
[build]   Abandoned: 10 packages were abandoned.                                                                                                                                                           
[build]   Failed:    8 packages failed.                                                                                                                                                                    
[build] Runtime: 2.3 seconds total. `

My default python version is 3.6.6. Even after I changed it to version  2.7.12, I got the same error.

So, do you know how to fix it?

johannes-graeter, 2019-02-19:
Hi, thanks for you interest in my work. 
Next week I will have my PhD defence, after that I will have time to cope with the error :)
Thanks for you patience!

OnionLuoJiayi, 2019-03-27:
Hi~ I have met similar problem. Have you figure it out yet?
I really want to try it. Please

poggenhans, 2019-03-27:
Can you upate `rosinterface_handler` and test again? We started fixing the python3 support of `rosinterface_handler`.

The issue was that after changing from `python3` to `python2` you have to clean and rebuild the whole workspace so that the changes have effect.

closed 2019-06-05 by johannes-graeter