johannes-graeter/limo #19: how to do the reconstruction from limo

wenjunj, 2019-03-07:
It is a great work about lidar-camera slam,that I want to know how to do the reconstruction,I watch the video in the youtube where there is a picture about reconstruction.Thank you

johannes-graeter, 2019-03-07:
Thank you :) It is a simple demonstration of the precision of the motion estimate. Simply take the LIDAR pointclouds, add the RGB values from the camera and transform the clouds in one coordinate system with the estimated motion.

wenjunj, 2019-03-07:
Thank you very much：），another question，have you experimented your algorithm in real environment？and how to do the calibration between camera and lidar，toolbox？

johannes-graeter, 2019-03-07:
Yes as part of my PhD I implemented it on our test vehicles and tested it running while driving :) Lidar-CAmera claibration is done by this method https://www.mrt.kit.edu/z/publ/download/2018/Automatic_Calibration_of_Multiple_Cameras_and_Depth_Sensors_with_a_Spherical_Target.pdf

wenjunj, 2019-03-07:
It is great！Thank you again! Can I appreciate your PhD thesis for detail？if OK，please send me by email （wjjiang@sjtu.edu.cn）Thank you~

johannes-graeter, 2019-03-07:
IT will be publsihed soon, I will then send it to you :)

wenjunj, 2019-03-21:
Hi : )could you please send the source code of demonstration about reconstruction from limo? I just want to      
 see the whole pointcloud map after reconstruction! Thank you !    email: wjjiang@sjtu.edu.cn

johannes-graeter, 2019-03-22:
I will have a look into that (usually I do not distribute messy code like that :) )

wenjunj, 2019-03-22:
Thank you so much, I will look forward to your email : )

johannes-graeter, 2019-04-09:
In limo/pointcloud plotting you find the python scripts with which you can accumulate the pointlocuds. accumulate_lidar_pcl_from_trajectory_estimation.py does the accumulation, vtk_viewer helps you plot it. Have fun :)

wenjunj, 2019-04-10:
/home/graeter/kitti_upload_candidates/likey_result_uploaded_rank13/{}.txt".format(sequence)
"/mrtstorage/datasets/kitti/odometry/complete/
could you please explain these two directory or release your dataset :) Thank you!

johannes-graeter, 2019-04-10:
Those are the parameters of the script. I added it as argparse arguments now, is it better understandable? Please try it, I am on the move at the moment and have no opportunity to test it...

wenjunj, 2019-04-14:
：）Have you run your script? I cannot run the script,and I find there are some changes when you use pykitti. Could you please give me a Tutorial to run the script accumulate_lidar_pcl_from_trajectory_estimation.py

wenjunj, 2019-04-14:
Thank you ！I have recieve the reconstruction,you use the true pose of kitti ?what about your code?
and the reconstruction result is not worse than your youtube video, what is the problem，Do I need to set parameter！thank you:)
[image]

johannes-graeter, 2019-04-14:
Hi,
no of course I use the estimated trajectory of LIMO for the video. I only use the LIDAR and camera data from Kitti, not the pose data.
The poses are loaded from the results textfile as written in the help.

Your sentence is confusing are the results worse or not?
I can't see anything on the screenshot you provided, the scale is too small, and the lighting of your viewer is very dark.
1. In order to see the quality I would need a better view, perhaps you can upload the ply file?
2. You use meshlab for viewing right? try the vtk based viewer I uploaded, it looks better.
3. As mentioned in other issues, the code provided here is older than the one with which I did the videos, however the quality should not differ so much, that it would be visible in the reconstruction.

Regards,

Johannes

closed 2019-06-05 by johannes-graeter