johannes-graeter/limo #23: how to run catkin run_tests

guan4286, 2019-03-31:


guan4286, 2019-03-31:
g@g:~/tutorial_ws/src/limo$ catkin run_tests --this --profile limo_release
==> Expanding alias 'run_tests' from 'catkin run_tests --this --profile limo_release' to 'catkin build --verbose --catkin-make-args run_tests -- --this --profile limo_release'
[build] Error: In order to use --this, the current directory must be part of a catkin package.

guan4286, 2019-03-31:
what does that mean? how shoule i do to solve this?

johannes-graeter, 2019-03-31:
Hi there, 
thanks for the interest in my work.
As the error suggests, your directory is not regcognized as catkin package folder.
Did you do the catkin init?
Did the packages build correctly?
If yes, try and go limo/keyframe_bundle_adjustment and run the tests there and report here :) 
Thanks!

Regards,

Johannes

guan4286, 2019-04-01:
here I have a question ,what is the commander "--this" for ?
I try it again like this :
1.catkin init      this step is normal .
2. catkin run_tests --this --profile limo_release    this step have error as before.so I remove "--this" as your advice,then i get the bellow result .

ibgtest.so CMakeFiles/gtest.dir/src/gtest-all.cc.o  -L/home/g/tutorial_ws/build_limo_release/velodyne_laserscan/gtest/gtest/src -lpthread -Wl,-rpath,/home/g/tutorial_ws/build_limo_release/velodyne_laserscan/gtest/gtest/src 
make[3]: Leaving directory '/home/g/tutorial_ws/build_limo_release/velodyne_laserscan'
[ 20%] Built target gtest
/usr/bin/make -f tests/CMakeFiles/test_system_node.dir/build.make tests/CMakeFiles/test_system_node.dir/depend
make[3]: Entering directory '/home/g/tutorial_ws/build_limo_release/velodyne_laserscan'
cd /home/g/tutorial_ws/build_limo_release/velodyne_laserscan && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/g/tutorial_ws/src/smartcar/driver/velodyne/velodyne_laserscan /home/g/tutorial_ws/src/smartcar/driver/velodyne/velodyne_laserscan/tests /home/g/tutorial_ws/build_limo_release/velodyne_laserscan /home/g/tutorial_ws/build_limo_release/velodyne_laserscan/tests /home/g/tutorial_ws/build_limo_release/velodyne_laserscan/tests/CMakeFiles/test_system_node.dir/DependInfo.cmake --color=
Dependee "/home/g/tutorial_ws/build_limo_release/velodyne_laserscan/tests/CMakeFiles/test_system_node.dir/DependInfo.cmake" is newer than depender "/home/g/tutorial_ws/build_limo_release/velodyne_laserscan/tests/CMakeFiles/test_system_node.dir/depend.internal".
Dependee "/home/g/tutorial_ws/build_limo_release/velodyne_laserscan/tests/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/home/g/tutorial_ws/build_limo_release/velodyne_laserscan/tests/CMakeFiles/test_system_node.dir/depend.internal".
Scanning dependencies of target test_system_node
make[3]: Leaving directory '/home/g/tutorial_ws/build_limo_release/velodyne_laserscan'
/usr/bin/make -f tests/CMakeFiles/test_system_node.dir/build.make tests/CMakeFiles/test_system_node.dir/build
make[3]: Entering directory '/home/g/tutorial_ws/build_limo_release/velodyne_laserscan'
[ 30%] Building CXX object tests/CMakeFiles/test_system_node.dir/system.cpp.o
cd /home/g/tutorial_ws/build_limo_release/velodyne_laserscan/tests && /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"velodyne_laserscan\" -I/home/g/tutorial_ws/devel_limo_release/.private/velodyne_laserscan/include -I/opt/ros/kinetic/include -I/opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp  -Wall -O2 -g -DNDEBUG   -o CMakeFiles/test_system_node.dir/system.cpp.o -c /home/g/tutorial_ws/src/smartcar/driver/velodyne/velodyne_laserscan/tests/system.cpp
/usr/bin/make -f tests/CMakeFiles/test_lazy_subscriber_nodelet.dir/build.make tests/CMakeFiles/test_lazy_subscriber_nodelet.dir/depend
make[3]: Entering directory '/home/g/tutorial_ws/build_limo_release/velodyne_laserscan'
cd /home/g/tutorial_ws/build_limo_release/velodyne_laserscan && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/g/tutorial_ws/src/smartcar/driver/velodyne/velodyne_laserscan /home/g/tutorial_ws/src/smartcar/driver/velodyne/velodyne_laserscan/tests /home/g/tutorial_ws/build_limo_release/velodyne_laserscan /home/g/tutorial_ws/build_limo_release/velodyne_laserscan/tests /home/g/tutorial_ws/build_limo_release/velodyne_laserscan/tests/CMakeFiles/test_lazy_subscriber_nodelet.dir/DependInfo.cmake --color=
Dependee "/home/g/tutorial_ws/build_limo_release/velodyne_laserscan/tests/CMakeFiles/test_lazy_subscriber_nodelet.dir/DependInfo.cmake" is newer than depender "/home/g/tutorial_ws/build_limo_release/velodyne_laserscan/tests/CMakeFiles/test_lazy_subscriber_nodelet.dir/depend.internal".
Dependee "/home/g/tutorial_ws/build_limo_release/velodyne_laserscan/tests/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/home/g/tutorial_ws/build_limo_release/velodyne_laserscan/tests/CMakeFiles/test_lazy_subscriber_nodelet.dir/depend.internal".
Scanning dependencies of target test_lazy_subscriber_nodelet
make[3]: Leaving directory '/home/g/tutorial_ws/build_limo_release/velodyne_laserscan'
/usr/bin/make -f tests/CMakeFiles/test_lazy_subscriber_nodelet.dir/build.make tests/CMakeFiles/test_lazy_subscriber_nodelet.dir/build
make[3]: Entering directory '/home/g/tutorial_ws/build_limo_release/velodyne_laserscan'
[ 40%] Building CXX object tests/CMakeFiles/test_lazy_subscriber_nodelet.dir/lazy_subscriber.cpp.o
cd /home/g/tutorial_ws/build_limo_release/velodyne_laserscan/tests && /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"velodyne_laserscan\" -I/home/g/tutorial_ws/devel_limo_release/.private/velodyne_laserscan/include -I/opt/ros/kinetic/include -I/opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp  -Wall -O2 -g -DNDEBUG   -o CMakeFiles/test_lazy_subscriber_nodelet.dir/lazy_subscriber.cpp.o -c /home/g/tutorial_ws/src/smartcar/driver/velodyne/velodyne_laserscan/tests/lazy_subscriber.cpp
/usr/bin/make -f tests/CMakeFiles/test_lazy_subscriber_node.dir/build.make tests/CMakeFiles/test_lazy_subscriber_node.dir/depend
make[3]: Entering directory '/home/g/tutorial_ws/build_limo_release/velodyne_laserscan'
cd /home/g/tutorial_ws/build_limo_release/velodyne_laserscan && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/g/tutorial_ws/src/smartcar/driver/velodyne/velodyne_laserscan /home/g/tutorial_ws/src/smartcar/driver/velodyne/velodyne_laserscan/tests /home/g/tutorial_ws/b [... cut ...]    
[build]   Failed:    2 packages failed.                                                                                                                                                                    
[build] Runtime: 2 minutes and 12.2 seconds total.                                                                                                                                                         
[build] Note: Workspace packages have changed, please re-source setup files to use them.

johannes-graeter, 2019-04-01:
catkin run_tests will execute all tests available. 
DO you have toher packages in the same workspace. For example  loam_velodyne which throws an errors here is not part of my project. Please make a clean workspace and try again. If that doesn'T work try with the virtual machine I supply (see README.md)

guan4286, 2019-04-02:
here i have a question about the paper.which is the part of the data fusion about laser and visual ?
I don't understand.please help me !

guan4286, 2019-04-03:
hi I'm here again.
 Do you have any other learning resources about visual and laser fusion ?
Can you recommend some for me?

chenxiaocongAI, 2019-04-22:
hello,I am as a student,I also want to know the about visual and laser fusion .Can you tell me your email.My email is 348733501@qq.com

closed 2019-04-30 by johannes-graeter