johannes-graeter/limo #25: [build] Error: In order to use --this, the current directory must be part of a catkin package.

Atian971, 2019-04-01:
Hello！ Many thanks for your project , when I'm trying your command"catkin run_tests --this --profile limo_release" ,I received the error as the title, so what does "this" means?
Besides, I am doing on my graduation project using rplidar and monocular camera, any ideas how to transplant it into my project?Thanks a lot

johannes-graeter, 2019-04-01:
<Thanks for the interest in my project.
Hm that problem seems to have popped up recently, see #23 .
Could you try my suggestions and tell me if it worked?
--this means, that only the current package is run. You can also try without the --this flag.

rplidar is that LiDar here? https://www.robotshop.com/de/de/rplidar-a2m8-360-laserscanner.html
I dont know that sensor, does it only have 1 layer? In that case it will be a bit hard to use LIMO as it is, since we need 3d information for the depth interpolation. 
However you could try to use dense optical flow (f.e. from PWC-Net, https://github.com/djl11/PWC_Net_TensorFlow) and track the points of the 1 scan line. From the point of having tracked image feature points and corresponding depth you can use the code in this repo to do the bundle adjustment.
I always wanted to try that but never had the time. 
If you have the tracklets with depth, write an issue and I will help you integrate the measurements with limo :)

Cheers,

Johannes

Atian971, 2019-04-04:
Thanks for your reply.I tried your idea that went into the Keyframe_bundle_adjustment folder and run again, it got better but still had some problems as belows:

[build] Summary: 10 of 12 packages succeeded.                                  
[build]   Ignored:   10 packages were skipped or are blacklisted.              
[build]   Warnings:  8 packages succeeded with warnings.                       
[build]   Abandoned: 1 packages were abandoned.                                
[build]   Failed:    1 packages failed.
The process was a bit long,so would you please tell me your email address and I get a screenshot to you?
I don't understand what your "layer"means,the rplidar is an 2d lidar with 5 lines,if it cannot offer depth information, what about fusing it with the depth camera "kinect"? 

Hm I am a beginner on slam,so sorry for my lack of knowledge

johannes-graeter, 2019-04-08:
Yeah the output is long, since I print out some results of the unittests.
You can upload the screenshot here(so others can profit from it as well.)

Is the LiDAR the one from Slamtec (http://www.slamtec.com/en/Lidar/A3)?
Some LiDARs have only 1 emitter and receiver, so the only have one plane in which they scan (what you refer to as 2d scanner), some as (https://velodynelidar.com/vlp-16.html) have several of those scanning lines, giving 3d measurements.

What I forgot: Did you source your workspace before running the tests? (command: source ${your_catkin_workspace}/devel/setup.sh)

johannes-graeter, 2019-04-23:
Hi I updated the repos, can you pull all (or clone anew) and try again?

closed 2019-06-26 by johannes-graeter