johannes-graeter/limo #27: timestamp problem

wenjunj, 2019-04-15:
Hi：）
How to solve the timestamp problem between camera and lidar ?
What specific camera and lidar do you use in the real environment?
Thank you

johannes-graeter, 2019-04-15:
Hi there,

in the real environment setup we use a blackfly with KOWA lenses with a view range of 100°.
LIDAR is the Velodyne HDL64.

Time synchronization is as a matter of fact not trivial.
In our case, however, the distortion of the "rolling shutter effect" of the LIDAR is not so important, since we only use local planes for depth extraction and we can use the "raw" pointclouds that are distorted.
Perhaps result would get even better, if you would use an IMU for undistortion(or buy a LiDAR that already does that for you).

Regards,

Johannes

closed 2019-04-15 by johannes-graeter

wenjunj, 2019-05-22:
Hi：）
how  do you to do the Time synchronization ? I have a velodyne HDL16 and a basler camera, and I use the  Time Synchronizer in ros ,the time interval between  velodyne and  camera is about 10-20ms. 
if you use Hardware synchronization，please teach me how to do that as detail  as possible!
Thank you :)