johannes-graeter/limo #29: catkin run_test error

14212094, 2019-06-03:
@johannes-graeter **Thank you for your great job!
When i run "catkin run_tests --profile limo_release"  command, i got this errror :**
test_results/keyframe_bundle_adjustment/gtest-keyframe_bundle_adjustment-keyframe_bundle_adjustment-test.xml: 16 tests, 0 errors, 1 failures, 0 skipped
Summary: 16 tests, 0 errors, 1 failures, 0 skipped
CMakeFiles/_run_tests_keyframe_bundle_adjustment.dir/build.make:57: recipe for target '_run_tests_keyframe_bundle_adjustment' failed
make[3]: Leaving directory '/home/liwenyi/limo_ws/build_limo_release/keyframe_bundle_adjustment'
CMakeFiles/Makefile2:270: recipe for target 'CMakeFiles/_run_tests_keyframe_bundle_adjustment.dir/all' failed
make[2]: Leaving directory '/home/liwenyi/limo_ws/build_limo_release/keyframe_bundle_adjustment'
make[3]: *** [_run_tests_keyframe_bundle_adjustment] Error 1
make[2]: *** [CMakeFiles/_run_tests_keyframe_bundle_adjustment.dir/all] Error 2
CMakeFiles/Makefile2:407: recipe for target 'CMakeFiles/run_tests.dir/rule' failed
make[1]: Leaving directory '/home/liwenyi/limo_ws/build_limo_release/keyframe_bundle_adjustment'
make[1]: *** [CMakeFiles/run_tests.dir/rule] Error 2
Makefile:303: recipe for target 'run_tests' failed
make: *** [run_tests] Error 2
cd /home/liwenyi/limo_ws/build_limo_release/keyframe_bundle_adjustment; catkin build --get-env keyframe_bundle_adjustment | catkin env -si  /usr/bin/make run_tests --jobserver-fds=6,7 -j; cd -

I am a newer to lidar_camera_slam.How can I solve this problem? Thank you so much!

johannes-graeter, 2019-06-05:
Hi,

I will have a look at it and give feedback (latest end of the week).

Regards,

Johannes

johannes-graeter, 2019-06-26:
Sorry i will need more time i am currently maintaining the project as a hobby:)

Brain97, 2019-10-22:
> @johannes-graeter **Thank you for your great job! When i run "catkin run_tests --profile limo_release" command, i got this errror :** [...]

HI， I  met the same problem. Did you solve this problem?

johannes-graeter, 2019-11-03:
I there, no i didn‘t it is very hard for me to maintain the code in my freetime since it takes a lot of time... don‘t know when i will be able to fix this , sorry...

zcg3648806, 2019-11-12:
also catkin run_test error:
---------------------------------------------------------------------------------------------------------------------
Errors     << keyframe_bundle_adjustment:make /home/zcg/catkin_limo/logs_limo_release/keyframe_bundle_adjustment/build.make.005.log

Failed     << keyframe_bundle_adjustment:make                     [ Exited with code 2 ]
Failed    <<< keyframe_bundle_adjustment                          [ 0.9 seconds ]
Abandoned <<< image_preproc_ros_tool                              [ Unrelated job failed ]
Abandoned <<< matches_conversion_ros_tool                         [ Unrelated job failed ]
Abandoned <<< util_nodes_tf2_ros_tool                             [ Unrelated job failed ]
Abandoned <<< tracklets_depth_ros_tool                            [ Unrelated job failed ]
Abandoned <<< viso_feature_tracking_ros_tool                      [ Unrelated job failed ]
Abandoned <<< keyframe_bundle_adjustment_ros_tool                 [ Unrelated job failed ]
Abandoned <<< demo_keyframe_bundle_adjustment_meta                [ Unrelated job failed ]

[ERROR] [1573524501.794911464]: Parameter '/test/rosinterface_handler_test/int_param' is not defined.
------------------------------------------------------------------------------------------------------------------------
In fact, I don't know what effect this test has on actual use. The compilation of the previous step has passed. But when I ran the packet, I couldn't see the track of the three colors.

johannes-graeter, 2019-11-12:
This seems to be an error caused by changes in an external library (rosinterface) probably more work for porting:) would you be willing to fix it? Shouldn‘t be too bad...

johannes-graeter, 2020-03-28:
I can not reproduce the error.
Using the new docker container, I added to the repo, the unit tests run fine...
Perhaps you try the container?

closed 2020-03-28 by johannes-graeter