johannes-graeter/limo #31: build error in ROS melodic

auracle-net, 2019-07-06:
Hi. 
I have followed the wiki/readme and tried to install with your provided script. But i could not get it built. I have two systems with Ubuntu 18.04, ROS melodic, opencv3.4.4, libpcl-dev 1.8.1, ceres 1.14.0. Both of my systems show same error and start failing while building the util_nodes_tf2_ros_tool.  Please check the dump file. I hope you have some time to help me to resolve it. Thank you. 

Best regards
Kawser Jahan

[unittests_dump.txt](https://github.com/johannes-graeter/limo/files/3365021/unittests_dump.txt)
[image]

johannes-graeter, 2019-07-06:
Hi there,

The system is not yet ported to 18.04 and melodic but is only tested on 16.04 and kinetic. At the moment I am in the process of porting it but as you mentioned I don‘t have a lot of time:) if you want to help I would appreciate it contact me if that is the case than I can coordinate:) in the mean time you can try it out with the virtual machine I have for download in the readme ( with 16.04 and kinetic)

johannes-graeter, 2019-11-03:
Sorry no update yet maintaining this in my freetime is quite hard :) if you would be able to help this would be much appreciated:) (write me if this is the case:)

johannes-graeter, 2019-12-23:
I ported it to 18.04+melodic (branch port_18.04). However there is a linker error originating from opencv extras. This seems to be a proble with the standard opencv versions from melodic and bionic.A solution is to build opencv yourself. 
I suspect that the problem will be solved with the new  ubuntu version that comes in Mai

closed 2019-12-23 by johannes-graeter