johannes-graeter/limo #33: Build Error..

LongruiDong, 2019-07-16:
Hi,
me again ~
after solved my last issue, i got this new Error
```
[build] Found '23' packages in 0.0 seconds.                                                                       
[build] Updating package table.                                                                                   
Warning: generated devel space setup files have been deleted.
Starting  >>> catkin_tools_prebuild                                                                               
__________________________________________________________________________________________________________________
Errors     << catkin_tools_prebuild:cmake /home/dlr/catkin_ws/logs_limo_release/catkin_tools_prebuild/build.cmake.000.log
CMake Error: The source directory "/home/dlr/catkin_ws/build_limo_release/catkin_tools_prebuild/-Werror=volatile-register-var" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
cd /home/dlr/catkin_ws/build_limo_release/catkin_tools_prebuild; catkin build --get-env catkin_tools_prebuild | catkin env -si  /home/dlr/anaconda3/envs/py27/bin/cmake /home/dlr/catkin_ws/build_limo_release/catkin_tools_prebuild --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/dlr/catkin_ws/devel_limo_release/.private/catkin_tools_prebuild -DCMAKE_INSTALL_PREFIX=/home/dlr/catkin_ws/install_limo_release -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_CXX_FLAGS=-Wall -Wextra -Wno-unused-parameter -Werror=address -Werror=array-bounds=1 -Werror=bool-compare -Werror=comment -Werror=enum-compare -Werror=format -Werror=init-self -Werror=logical-not-parentheses -Werror=maybe-uninitialized -Werror=memset-transposed-args -Werror=nonnull -Werror=nonnull-compare -Werror=openmp-simd -Werror=parentheses -Werror=return-type -Werror=sequence-point -Werror=sizeof-pointer-memaccess -Werror=switch -Werror=tautological-compare -Werror=trigraphs -Werror=uninitialized -Werror=volatile-register-var; cd -
..................................................................................................................
Failed     << catkin_tools_prebuild:cmake                         [ Exited with code 1 ]                          
Failed    <<< catkin_tools_prebuild                               [ 0.3 seconds ]                                 
Abandoned <<< aloam_velodyne                                      [ Unrelated job failed ]                        
Abandoned <<< mrt_cmake_modules                                   [ Unrelated job failed ]                        
Abandoned <<< image_preproc                                       [ Unrelated job failed ]                        
Abandoned <<< matches_msg_depth_ros                               [ Unrelated job failed ]                        
Abandoned <<< matches_msg_ros                                     [ Unrelated job failed ]                        
Abandoned <<< matches_msg_types                                   [ Unrelated job failed ]                        
Abandoned <<< matches_msg_conversions_ros                         [ Unrelated job failed ]                        
Abandoned <<< monolidar_fusion                                    [ Unrelated job failed ]                        
Abandoned <<< robust_optimization                                 [ Unrelated job failed ]                        
Abandoned <<< keyframe_bundle_adjustment                          [ Unrelated job failed ]                        
Abandoned <<< rosinterface_handler                                [ Unrelated job failed ]                        
Abandoned <<< image_preproc_ros_tool                              [ Unrelated job failed ]                        
Abandoned <<< matches_conversion_ros_tool                         [ Unrelated job failed ]                        
Abandoned <<< util_nodes_tf2_ros_tool                             [ Unrelated job failed ]                        
Abandoned <<< viso2                                               [ Unrelated job failed ]                        
Abandoned <<< feature_tracking_core                               [ Unrelated job failed ]                        
Abandoned <<< tracklets_depth_ros_tool                            [ Unrelated job failed ]                        
Abandoned <<< viso_feature_tracking_ros_tool                      [ Unrelated job failed ]                        
Abandoned <<< keyframe_bundle_adjustment_ros_tool                 [ Unrelated job failed ]                        
Abandoned <<< demo_keyframe_bundle_adjustment_meta                [ Unrelated job failed ]                        
[build] Summary: 0 of 21 packages succeeded.                                                                      
[build]   Ignored:   3 packages were skipped or are blacklisted.                                                  
[build]   Warnings:  None.                                                                                        
[build]   Abandoned: 20 packages were abandoned.                                                                  
[build]   Failed:    1 packages failed.
```
my environment:
```
ubuntu 16.04  ROS kinetic
python 2.7
```
Can you help me solve this error?
Thanks in advance!!

daaakong, 2019-07-23:
I have the same problem....

johannes-graeter, 2019-11-03:
Hi there,
catkin tools is not maintained by me.
My best guess is that they changed the library structure. In my spare time I am porting limo to melodic and 18.04 however this is quite hard on the side of my job. I would be happy if you would be willing to help me and fix this issue? Best

johannes-graeter, 2020-03-28:
Now I added a docker container that builds.
Happy coding :)

closed 2020-03-28 by johannes-graeter