johannes-graeter/limo #37: I can‘t download the data sets! Can you give me another links?

Bill-Deng, 2019-12-02:
404 not found
The requested URL was not found on this server.

Bill-Deng, 2019-12-02:
google driver is OK

johannes-graeter, 2019-12-02:
Hi, I noticed that yesterday night, my enployee, the university must have taken down my account. I will need to get some virtual space, google drive is too small unfortunately, since the files habe several GB

Bill-Deng, 2019-12-02:
Thanks，can you upload 04.bag (2.05GB) or 01.bag(7.78GB) to the google drive? I know 00.bag is too big.

Bill-Deng, 2019-12-06:
Hello？could you please upload 04.bag.Google drive has 15 GB free spaces!!

hcyoo93, 2019-12-17:
Is there anyone who has the bag file? I have same issue and unfortunately it is urgent..

johannes-graeter, 2019-12-17:
Trying to upload i ran into problems with my hard drive and need to restore it. Hope to be able to solve it this week.

johannes-graeter, 2019-12-23:
The data is corrupt, I will need to get it from the Insitutes server.
Sorry for the wait.

Bill-Deng, 2019-12-30:
johannes，how did you make the bag file(especially at part of segmentation),did you have the tools or use the kitti's tools?

johannes-graeter, 2019-12-31:
Hi,
I had my own tool, which really was nothing more than using rosbag api to write data into a bag file.
The part which was a bit more difficult was adding calibration data as as static transformations with tf2, since I had to manipulate latching topics.
Unfortunately I do not have the rights to open source that... But also the tool was quite hacky...

There is a tool for converting kitti to rosbag from eth https://github.com/ethz-asl/kitti_to_rosbag but as far as I know it does only support the format from kitti raw (probably it would be not too much if big a deal to contirbute an odometry dataset format though...) and for compatibility with my code calibration must be supplied as tf2_static which is not the case (and kind of a hazle to implement)
A neat python tool is this one: https://github.com/utiasSTARS/pykitti 
However you would need to wirte the rosbag yourself with the python rosbag api and also fix the issue with static calibrations...

For semantics I used semantic segmentation form our institute, based on segnet.
Ther is a whole lot of semantic segmentation networks out there https://github.com/mrgloom/awesome-semantic-segmentation . I think any of them would do. However make sure that the labels are correctly mapped to outlier labels and shrubbery(trees, bushes, grass,...)(https://github.com/johannes-graeter/limo/blob/master/keyframe_bundle_adjustment_ros_tool/res/outlier_labels.yaml)

Happy new year 2020

Bill-Deng, 2020-01-02:
Thanks very much，it's really helpful，happy new year！！

johannes-graeter, 2020-02-09:
Hi there,

Uploaded sequneces 01.bag and 04.bag to google drive (see README) happy coding: :)

Best, Johannes

closed 2020-02-09 by johannes-graeter