johannes-graeter/limo #41: Fail to run with own bag file

hcyoo93, 2019-12-18:
Hello,

I made a bag file with the package in eth repo https://github.com/ethz-asl/kitti_to_rosbag

I used, kitti processed data on raw data page not the odometry dataset! 

Because, the kitti_to_rosbag package is optimized for the raw dataset.

`types:       sensor_msgs/CameraInfo  [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/Image       [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181]
topics:      /sensor/camera/color/left/camera_info       4544 msgs    : sensor_msgs/CameraInfo 
             /sensor/camera/color/left/image_rect        4544 msgs    : sensor_msgs/Image      
             /sensor/camera/grayscale/left/camera_info   4544 msgs    : sensor_msgs/CameraInfo 
             /sensor/camera/grayscale/left/image_rect    4544 msgs    : sensor_msgs/Image      
             /sensor/velodyne/cloud_euclidean            4544 msgs    : sensor_msgs/PointCloud2
`
I set topic name and frame ID as you did and then used static transform publisher in launch file to organize /sensor/camera/grayscale/image_rect, /sensor/camera, sensor/velodyne.

However, when I tried to run the code it fails with

`Publisher name calc depth stats: /monolidar_debug/depthcalc_stats
Publisher name tracklets depth with depth: /tracklets_depth/tracklets/left
[ INFO] [1576654127.668801487, 1317614134.932880896]: Waiting for the ROSBAG to start
**[ERROR] [1576654127.668993070, 1317614134.932880896]: "sensor/camera" passed to lookupTransform argument target_frame does not exist.** 
[ INFO] [1576654129.236735610, 1317614134.953554959]: waitForService: Service [/feature_matching_nodelet_manager/load_nodelet] is now available.
[ INFO] [1576654129.257601309, 1317614134.955565575]: Started '/feature_matching_nodelet_manager' in namespace '/'.
Subscribed topics: 
/clock
/sensor/camera/grayscale/left/image_rect_g
Advertised topics: 
/rosout
/matches/grayscale/left

[ INFO] [1576654129.539362769, 1317614134.983783477]: Started '/mono_lidar' in namespace '/'.
Subscribed topics: 
/clock
/tf
/tf_static
/tracklets_depth/tracklets/left/with_outlier_dummy
/sensor/camera/grayscale/left/camera_info
Advertised topics: 
/rosout
/mono_lidar/parameter_descriptions
/mono_lidar/parameter_updates
/tf
/diagnostics
/estimate/active_path
/estimate/complete_path
/estimate/landmarks
/mono_lidar/planes_topic

[ INFO] [1576654130.129828192, 1317614135.042285784]: /gamma_correction_nodelet_left Gamma correction Entered callback!
[ INFO] [1576654130.135412008, 1317614135.043293510]: /gamma_correction_nodelet_left gamma correction leaving callback!
[ INFO] [1576654130.207103483, 1317614135.050340213]: Duration feature matching and tracking: 70 ms
[ INFO] [1576654131.164227897, 1317614135.146308676]: /gamma_correction_nodelet_left Gamma correction Entered callback!
[ INFO] [1576654131.166908270, 1317614135.146308676]: /gamma_correction_nodelet_left gamma correction leaving callback!
[ INFO] [1576654131.334547752, 1317614135.163416503]: Duration feature matching and tracking: 167 ms
[ INFO] [1576654132.201558240, 1317614135.249148301]: /gamma_correction_nodelet_left Gamma correction Entered callback!
[ INFO] [1576654132.204351554, 1317614135.250195713]: /gamma_correction_nodelet_left gamma correction leaving callback!
[ INFO] [1576654132.371910776, 1317614135.266305882]: Duration feature matching and tracking: 167 ms
[ INFO] [1576654133.237917031, 1317614135.353027152]: /gamma_correction_nodelet_left Gamma correction Entered callback!
[ INFO] [1576654133.240097067, 1317614135.354035724]: /gamma_correction_nodelet_left gamma correction leaving callback!
[ INFO] [1576654133.405876227, 1317614135.370135368]: Duration feature matching and tracking: 165 ms
[ INFO] [1576654134.037473093, 1317614135.433676594]: Got the tf transformations: 
-0.00185775   -0.999966 -0.00803998 -0.00478403
-0.00648148  0.00805186   -0.999947  -0.0733743
   0.999977 -0.00180554 -0.00649622   -0.333997
          0           0           0           1
[ INFO] [1576654134.071498188, 1317614135.436697895]: Got the camera info:
Got the camera info:
img_width: 1241
img_height: 376
focal_length_x: 718.856
principalPointX: 607.193
principalPointY: 185.216

Initialize DepthEstimator Post
DepthEstimator successful initialized
[ INFO] [1576654134.274471827, 1317614135.456825877]: /gamma_correction_nodelet_left Gamma correction Entered callback!
[ INFO] [1576654134.275445997, 1317614135.456825877]: /gamma_correction_nodelet_left gamma correction leaving callback!
size before randomization 124606
size after randomization 6000
Publish image projection cloud
[ INFO] [1576654134.302916600, 1317614135.459845413]: Duration tracklet_depth_ros_tool=231 ms
[ INFO] [1576654134.438697818, 1317614135.473938552]: Duration feature matching and tracking: 163 ms
size before randomization 124479
size after randomization 6000
Publish image projection cloud
[ INFO] [1576654134.829428427, 1317614135.512193603]: Duration tracklet_depth_ros_tool=522 ms
terminate called after throwing an instance of 'std::runtime_error'
  what():  Time is out of dual 32-bit range
================================================================================**REQUIRED process [mono_lidar-19] has died!
process has died [pid 6420, exit code -6, cmd /home/hcyoo/catkin_ws6/devel/lib/keyframe_bundle_adjustment_ros_tool/mono_lidar __name:=mono_lidar __log:=/home/hcyoo/.ros/log/4b02f8ea-20d1-11ea-854d-74d4358c6f8e/mono_lidar-19.log].
log file: /home/hcyoo/.ros/log/4b02f8ea-20d1-11ea-854d-74d4358c6f8e/mono_lidar-19*.log
Initiating shutdown!**
================================================================================
`
Also, I found that **feature_matching_nodelet_manager** does not subscribe sensor/camera/grayscale/left. but I cannot figure out further.

Note that, time sync is slightly different ~0.00x sec (because this is not odometry dataset there has some time difference)

Does anyone have same experience?

Please give me some comments

Thank you!

johannes-graeter, 2019-12-23:
Hi,

Thanks for trying this with your own data.
I think the problem might be here:
 Duration tracklet_depth_ros_tool=528 ms
terminate called after throwing an instance of 'std::runtime_error'
  what():  Time is out of dual 32-bit range

seems to me that the time stamps have the wrong format, does the converter from eth use ros::Time instances?

Unfortunately my data was corrupted during regeneration, I will try to get in contact with my old institute to have somebody send it to me...

Best, Johannes

johannes-graeter, 2020-02-09:
Hi there,

Uploaded sequneces 01.bag and 04.bag to google drive (see README) happy coding: :)

Best, Johannes

xuwuzhou, 2020-02-16:
@johannes-graeter 
Thanks for your work!
But I still have question about how to make a bag like your example bag,do you have any suggestions?
Best wishes!

johannes-graeter, 2020-03-28:
Hm this seems to be quite a bummer... 
Since I want to restart the development on this now during corona break, I think I will rework the interface so that standard kitti data can be used.

closed 2020-03-29 by johannes-graeter