johannes-graeter/limo #42: odometry

Bill-Deng, 2020-03-13:
where is the Odometry's code,I noticed that you did't use the libviso2.

Bill-Deng, 2020-03-13:
Did you use the BA

johannes-graeter, 2020-03-28:
Sorry for the late reply.
By odometry you mean a frame to frame motion estimate without using prior timesteps?(in my terms odometry estimate also includes a longer history. The whole methodology including BA would hence be called odometry since I do not use loop closure...)
If so:
viso2 is only for the feature matcher, the odometry is not used.
It can be used as an input to the BA node, but usually I save the time and use the prior estimate of the BA and linear interpolation to the future. 
In my dissertation however I show robustness gain by using frame to frame motion estimation from multi-camera surround setup, using https://github.com/johannes-graeter/momo. DIssertation is here https://books.google.de/books?hl=en&lr=&id=cZW8DwAAQBAJ&oi=fnd&pg=PR3&dq=info:bevZJ9ky3sYJ:scholar.google.com&ots=ZO8u7vNceo&sig=kmN0tlAUyZ0XM0XHrexIiK-61Zw&redir_esc=y#v=onepage&q&f=false (unfortunately only in german :) )

closed 2020-03-28 by johannes-graeter