johannes-graeter/limo #45: Image source when runtime.

thanhphong98, 2020-03-25:
Hello,
Recently, I try to build your package with my robot but I have many problems when simulation.
I don't know what mean of "src/image", "tgt/image", "tgt/matches", "src/image" in demo_keyframe_bundle_adjustment_meta/launch/feature_matching.launch
[image]
I try to pass data of grayscale camera to topic sensor/camera/grayscale/left/image_rect and rgb camera to topic sensor/camera/color/left/image_rect_g. After push information to terminal, I only get above information. It not similar when run your rosbag. I think there are no feature matching.
[image]
Can you help me to solve this problem.

johannes-graeter, 2020-03-25:
Hi there,

src/Image and tgt/Image are just intermediate topics of image preprocessing. The topic for the images is correct, however the feature matching takes way too long... it should be more in the range of 30 ms what is the resolution of your images. perhaps you could downscale them to kitti resolution? Did you compile it in release (O3)?

johannes-graeter, 2020-03-28:
Hi,

did you try with lower resolution, yet?

Best,

Johannes

thanhphong98, 2020-03-29:
Hi,
Thanks for your enthusiastic support.
I try with 480x360 resolution but nothing seem to work. But I think my
simulation has some problem. I only use 2 camera instead of 4 camera like
your car and it 's not stereo camera. My lidar is laser scanner, not 3D
lidar. I get data from it and convert to pointcloud2.
I only have 2d lidar and monocular camera because of lacking of funding so
I try to build your code with my car system. Do you think your package can
run with my hardware ?
Phong.

johannes-graeter, 2020-03-29:
I am honoured by people that invest time in trying to use my work and even file pull requests to my repo, so it is you that I must thank :)
Two cameras is not a problem, actually you only need one, either gray or color.
The 2d lidar scanner is a problem though: In the publication this is explained more in detail, but the base line is that at the current state you need lidar points around each successfull camera feature match. Then with an algorithm described in the paper the depth is extracted, mainly its a fancy form of interpolation on a plane. that this plane is stable however we need 3d points, with a line scanner no valid interpolation plane can be extracted.

If you are motivated to develop for this, you could write a simple node that replaces mono_lidar_depth that assigns depth of feature matches to nearest points lidar points.
I tried in a 3h session, but on KITTI it didn't work well, perhaps for your setup it is better?

I am reworking this part however. My goal is to be able to use as less scan points as possible, however I am not sure if it will meet low computing power limitations since I plan to use dense optical flow from machine learning...

Best, Johannes

johannes-graeter, 2020-03-29:
Before you dive into that, I guess we should find out why the feature matching is so slow.
AFter merging limo2 I will isolate modules in order to run benchmarks with the raw kitti data. I think that would be a first step to see where to start tuning.

Best,
Johannes

thanhphong98, 2020-04-01:
Hi,
I spent time considering your package and I know that I can't
continuous now because my hardware is not accordant. I hope that I can
implement your package to make it runtime with my robot. Now I think I will
integrated machine learning to my robot to complete thesis. This is many
available package to integrate quickly :v
I'm very honored that you spend more time with me.
Wish you all the best.
Phong.

closed 2020-06-08 by johannes-graeter