johannes-graeter/limo #47: Build with Ubuntu 16.04

yjsx, 2020-04-03:
Hello！
I'm very interested in limo, but I met some question when I tried to build limo. Can I still build limo with Ubuntu 16.04 now？I notice that you may be using ros-melodic now, and  when I tried to build, I met so much error with ros-kinetic. Part of them is showing in the picture.
[image]
[image]
[image]
Or could you give me some suggestion to solve them? 
Thank you very much!

johannes-graeter, 2020-04-03:
Hi there,

since most people where interested in melodic and kinetic seems kind of outdated, I switched to melodic per default.
There is two ways: 
1. You could checkout an older version of the repo (f.e. https://github.com/johannes-graeter/limo/commit/b6504b95743ef311646efdfd928da22233733814) but make sure, that you also checkout a commit around the same date for the other repos needed (listed in install_repos)

2. You can try out the brand new docker image (that still uses melodic though, but you could have it beside the other kinetic based elements you use.)

johannes-graeter, 2020-04-03:
Also if you just want to develop on VO, I am working on a python interface for limo right now.
A first draft should be ready by Sunday.

Best,

Johannes

closed 2020-06-08 by johannes-graeter