johannes-graeter/limo #48: eigen error in mono_lidar

magic428, 2020-06-08:
Hi, johannes,

Thanks for your work. I encounter the error blow when running `roslaunch demo_keyframe_bundle_adjustment_meta kitti_standalone.launch ` with  04.bag, what's the
 reason of this error?

`mInitialize DepthEstimator Post
DepthEstimator successful initialized
[ INFO] [1591597391.300192954, 0.520561176]: /gamma_correction_nodelet_left Gamma correction Entered callback!
[ INFO] [1591597391.301096608, 0.520561176]: /gamma_correction_nodelet_left gamma correction leaving callback!
[ INFO] [1591597391.460237276, 0.536702683]: Duration feature matching and tracking: 158 ms
size before randomization 124232
size after randomization 6000
Publish image projection cloud
[ INFO] [1591597391.503006156, 0.540724514]: Duration assign semantic labels=0 ms
[ INFO] [1591597391.511459588, 0.541729991]: Duration tracklet_depth_ros_tool=376 ms
size before randomization 124088
size after randomization 6000
Publish image projection cloud
[ INFO] [1591597392.208998584, 0.611303525]: Duration assign semantic labels=2 ms
[ INFO] [1591597392.210731384, 0.611303525]: Duration tracklet_depth_ros_tool=696 ms
mono_lidar: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 16; int MatrixOrArrayOptions = 0]: Assertion `(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
================================================================================REQUIRED process [mono_lidar-20] has died!
process has died [pid 15517, exit code -6, cmd /home/magic/ros_wspace/limo_ws/devel/lib/keyframe_bundle_adjustment_ros_tool/mono_lidar __name:=mono_lidar __log:=/home/magic/.ros/log/8131c29a-a950-11ea-bc34-8c04ba998175/mono_lidar-20.log].
log file: /home/magic/.ros/log/8131c29a-a950-11ea-bc34-8c04ba998175/mono_lidar-20*.log
Initiating shutdown!
================================================================================
[ INFO] [1591597392.342114657, 0.624392962]: /gamma_correction_nodelet_left Gamma correction Entered callback!
[ INFO] [1591597392.343137421, 0.625400988]: /gamma_correction_nodelet_left gamma correction leaving callback!
`

johannes-graeter, 2020-06-08:
Hi magic428,

thanks for the interest in my work and reporting the bug!
This one is particularly nasty:
Eigen does a lot of smart stuff using aligned memroy in order to speed up matrix operations. However this means they do not interact well with stl containers and also struct declarations.
I circumvent this by deactivating the alignment in the CMakeLists.
Could you:
1. Tell me if you used the docker build or the normal build?
2. supply the gcc version you are compiling with?
3. if version > 7.0 could you try 
```bash
catkin config --profile limo_release -x _limo_release --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_CXX_FLAGS="-O3 -c++17"
```
( in the link they said that compiling with c++17 standard would solve the problem :)
4. Clean, build, rerun and tell me the result.

Best, Johannes

PS: The time usage for tracklet_depth_ros_tool is way too much, which is why I added the O3 option (letter not numeric ;) )

magic428, 2020-06-09:
Hi, Johannes,

thanks for the reply, and I have tried the "-std=c++17" that you suggested, but it didn't work.  

This is the compile Flags:

```bash
cd /home/magic/ros_wspace/limo/build_limo_release/demo_keyframe_bundle_adjustment_meta; catkin build --get-env demo_keyframe_bundle_adjustment_meta | catkin env -si  /usr/bin/cmake /home/magic/ros_wspace/limo/src/limo/demo_keyframe_bundle_adjustment_meta --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/magic/ros_wspace/limo/devel_limo_release/.private/demo_keyframe_bundle_adjustment_meta -DCMAKE_INSTALL_PREFIX=/home/magic/ros_wspace/limo/install_limo_release -DCMAKE_BUILD_TYPE=RelWithDebInfo '-DCMAKE_CXX_FLAGS=-O3 -std=c++17'; cd -
```

PS:

1. I use normal build instead of docker build.
2. My gcc(or g++) version:  7.5.0.

Best, magic428

magic428, 2020-06-09:
> Hi, Johannes,
> 
> thanks for the reply, and I have tried the "-std=c++17" that you suggested, but it didn't work.
> 
> This is the compile Flags:
> 
> ```shell
> cd /home/magic/ros_wspace/limo/build_limo_release/demo_keyframe_bundle_adjustment_meta; catkin build --get-env demo_keyframe_bundle_adjustment_meta | catkin env -si  /usr/bin/cmake /home/magic/ros_wspace/limo/src/limo/demo_keyframe_bundle_adjustment_meta --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/magic/ros_wspace/limo/devel_limo_release/.private/demo_keyframe_bundle_adjustment_meta -DCMAKE_INSTALL_PREFIX=/home/magic/ros_wspace/limo/install_limo_release -DCMAKE_BUILD_TYPE=RelWithDebInfo '-DCMAKE_CXX_FLAGS=-O3 -std=c++17'; cd -
> ```
> 
> PS:
> 
> 1. I use normal build instead of docker build.
> 2. My gcc(or g++) version:  7.5.0.
> 
> Best, magic428

johannes-graeter, 2020-06-09:
Hm i wonder why -DCMAKE_CXX in quotes...
Could you try to build it with the docker? In that way we can ensure to habe the same environment when I debug.

Best, Johannes

magic428, 2020-06-09:
> Hm i wonder why -DCMAKE_CXX in quotes...
> Could you try to build it with the docker? In that way we can ensure to habe the same environment when I debug.
> 
> Best, Johannes

Because -DCMAKE_CXX was set like this:  

```bash
catkin config --profile limo_release -x _limo_release --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_CXX_FLAGS="-O3 -std=c++17"
```

I will try to build in docker next. Thanks Johannes.

Best, magic428

johannes-graeter, 2020-06-28:
Hi magic,

did you try?
If it worked please write a short comment so i can close the issue :) 

Best, Johannes

magic428, 2020-06-29:
> Hi magic,
> 
> did you try?
> If it worked please write a short comment so i can close the issue :)
> 
> Best, Johannes

Hi, Johannes, 

Sorry for not reply you. I tried build in docker and it works, then I processed normal build successfully on another PC, So I think maybe some dependency broken or not compatible.  

Thanks for your work again! 

Best, magic428

closed 2020-06-29 by johannes-graeter