johannes-graeter/limo #49: Parametrs description

izzat1, 2020-06-12:
Thanks for sharing the code. Do you have a document describing in more detail the parameters used in the code.

johannes-graeter, 2020-06-12:
Hi,

thanks for the interest in my code.
The parameters are quite a lot. They are set in the launch files, that you find in the demo_keyframe... package one for the mono cam variant and one for lidar+camera. To get the whole parameterization you will need to go to the sub launch files of the components that are called there as well. 
Since all parameters are published on the rosparam server, the easiest way to see all at once is to run the ros node, pause the rosbag and have a look at the parameters with the rosparam gui (or command line).

Don‘t hesitate to ask further if you do not find it.

Best,
Johannes

izzat1, 2020-06-12:
Thanks for your feedback. Indeed the parameters are quite a lot. I will use your demo package as guidance and change some parameters to see its impact on performance.

closed 2020-06-12 by izzat1