johannes-graeter/limo #50: build error

hshsjkjy, 2020-06-24:
While running step 11/17 : RUN /bin/bash -c 'source /opt/ros/melodic/setup.bash && catkin build'
I met such a problem 
[image]
and I have already installed the Eigen
[image]

How can I solve this bug？

johannes-graeter, 2020-06-25:
Hi,

thanks for the interest in my work and the usage of the new docker container :)

Since the container does install libeigen there must be some changes in the links of eigen. In this issue https://stackoverflow.com/questions/23284473/fatal-error-eigen-dense-no-such-file-or-directory#25537967 there was a similar problem. I will need to debug it further, i will give it a spin this weekend :)

Thanks for your patience!
Best,
Johannes

hshsjkjy, 2020-06-26:
Thank you for your reply. I did create symlinks to solve the problem, but it doesn`t work. I wonder if I should re-install the libeigen.
Looking forward to your solution.

johannes-graeter, 2020-06-26:
Hi:)

did you do the link on your machine or in the Dockerfile (that is building the docker image).
please consider that you are building a docker image, so what you have installed (and linked) on your machine ist not relevant for the container (which is the beauty of it :) since you have an isolated system then) what you can do is add a RUN statement in the Dockerfile, that does the link to eigen and rebuild it :)

Thanks for your effort!
Best
Johannes

hshsjkjy, 2020-06-27:
Thanks for reminding me of that, now it works well by creating symlinks in the Dockerfile.
Sorry to disturb you for such a stupid bug, hope you have a good day:)

johannes-graeter, 2020-06-27:
> Thanks for reminding me of that, now it works well by creating symlinks in the Dockerfile.
> 
> Sorry to disturb you for such a stupid bug, hope you have a good day:)

No worries, there is no such thing as stupid bugs ;)
I am happy that you are interested in my code :)!
I would be happy if you would add it to the project as your contribution. Fork the repo make branch and file a pull request:) in that way your contrib is documented ;)

Best, Johannes

closed 2020-06-28 by johannes-graeter