johannes-graeter/limo #51: KITTI Ros bags

hlezki, 2020-06-28:
Hi,

Thank you for sharing your work. 
Can you also share other KITTI rosbags that are appropriate for your code? 

Thank you

johannes-graeter, 2020-06-28:
Hi,

thanks for the interest in my work!
Unfortunately the bags are very big and I do not possess more webspace :) However I am on the brink of finishing integration with a python interface that could read the dataset via pykitti. Help would be appreciated, are you interested?
Best,
Johannes

hlezki, 2020-06-28:
I have one more question. When I run your code with KITTI sequence-04, in the output poses file (/tmp/poses_dump.txt), i can not see poses for all images. In the groundtruth poses, there are 271 poses information for 271 images. Limo outputs only 112 poses for 112 images. Why is that? And how can I compare limo outputs with KITTI groudtruth ?? Is it possible to obtain same number pose information with the KITTI odometry groundtruth data?

johannes-graeter, 2020-06-28:
The ros version is built as an only system, so when your machine can not process the poses when they come in, they will be dropped. You can try running it slower with rosbag play -r 0.3 04.bag (i hope ingot the command right from the top if my head ;) )
By the way this is exactly why I wanted to add the pykitti interface going frame by frame :) 

Best,
Johannes

hlezki, 2020-06-28:
So  did you get your results that given in the paper by slowing down the rosbags for KITTI benchmark ?? And I wonder if i run rviz in the docker container?? I could not achieve running rviz in the docker. It seems it is not installed. How can I visualize the map with the RVIZ using your docker image? 

Thank you

johannes-graeter, 2020-06-28:
Hi hlezki,
 No the evaluation was in real time (we also used it on our autonomous vehicle in the field). This is described in the paper shortly and more kn depth in my phd thesis (the latter in german) but depending on your machine it might be necessary to slow it down. Especially if your RAM is small, the memory allocation in swap slows it down vastly. If speed is an issue further downscaling of the images could help (in my experience the results are a bit worse but not too much).
 The docker image was only added very recently, i did not intend to use rviz with it but rather serve the python interface. Probably you would need to setup a remote ros master and connect that with the docker, but i habe no experience with this. you can still build it with the bash script for rviz compatibility.

Best, Johannes

hlezki, 2020-06-28:
Hi,
Thank you for your comments. I am looking forward to your new improvements about adding the pykitti interface. 

Have a nice day.

johannes-graeter, 2020-06-28:
:) thx. Since I am working on this as a hobby, improvement is slower than I would like it to be. you are very welcome to contribute! Open source lives and dies with the contributions of the community!

Best,
Johannes

closed 2020-06-28 by johannes-graeter

Diggle221, 2020-10-07:
> Thanks for quick reply.
> I have one more question. When I run your code with KITTI sequence-04, in the output poses file (/tmp/poses_dump.txt), i can not see poses for all images. In the groundtruth poses, there are 271 poses information for 271 images. Limo outputs only 112 poses for 112 images. Why is that? And how can I compare limo outputs with KITTI groudtruth ?? Is it possible to obtain same number pose information with the KITTI odometry groundtruth data?

Excuse me,when I run  code with KITTI sequence-04 and 01, i can not find  the output poses file .Can you tell me where it is？Thank you very much.