johannes-graeter/limo #52: Odometry Evaluation Results

hlezki, 2020-06-29:
Hi, sorry to bother you again. 

As you suggested to me, I did slow down the rosbag speeds and I obtained same number of poses for each frame. However the results are not good. Can you please comment about, why I obtained such that results???
In the attachments, you can see the results for KITTI_01 and KITTI_04. 

Thanks in advanced. 

[image]
[image]

johannes-graeter, 2020-06-29:
Yes there seems something very wrong.
Slow speed can break things in real time slam. when you miss too much data, mapping part will be unstable. However, when you put it on really slow it should be ok...

It is hard to debug remotely though. You can send a dump of the verbose output that is printed on console, perhaps i will see sth there...
Also send me the specs of your machine...
You are using the docker right? 

Best,Johannes

hlezki, 2020-06-29:
Yes, I am using docker and I did really slow down the rosbag speed. Even I tried for different speeds such that -r 0.01 or -r 0.1 and -r 0.08.
In the following you can see the specs of my computer and terminal output.
Thank you 

[image]

  Residual evaluation                  0.0103
  Jacobian evaluation                  0.0671
  Linear solver                        0.0001
Minimizer                              0.0778

Postprocessor                          0.0000
Total                                  0.0779

Termination:                      CONVERGENCE (Function tolerance reached. |cost_change|/cost: 5.564529e-07 <= 1.000000e-06)

Duration solveTrimmed=0.129 sec

quaternion diff=0.00335895
[ INFO] [1593415566.005908192, 113.805810937]: In MonoLidar: number selected keyframes 0
[ INFO] [1593415566.005937287, 113.805810937]: In MonoLidar: number active keyframes 6
[ INFO] [1593415566.005957303, 113.805810937]: In MonoLidar: number active landmarks 637
[ INFO] [1593415566.005971019, 113.805810937]: In MonoLidar: number selected landmarks 445
[ INFO] [1593415566.062844538, 113.811860535]: 
Duration 5 point=4 ms
Duration pose only adjustment=132 ms
Duration select keyframes=0 ms
Duration push=0 ms
Duration publish=56 ms
Duration send pose tf=0 ms
time callback=0.196855 sec

[ INFO] [1593415566.196201111, 113.822998822]: /gamma_correction_nodelet_left Gamma correction Entered callback!
[ INFO] [1593415566.199486510, 113.825438660]: /gamma_correction_nodelet_left gamma correction leaving callback!
[ INFO] [1593415566.357686768, 113.841582900]: Duration feature matching and tracking: 157 ms
size before randomization 113409
size after randomization 6000
Publish image projection cloud
[ INFO] [1593415566.953469770, 113.901069085]: Duration assign semantic labels=1 ms
[ INFO] [1593415566.958866245, 113.901069085]: Duration tracklet_depth_ros_tool=599 ms
speed=1.46544
iter      cost      cost_change  |gradient|   |step|    tr_ratio  tr_radius  ls_iter  iter_time  total_time
   0  1.765585e+02    0.00e+00    3.87e+02   0.00e+00   0.00e+00  1.00e+04        0    1.18e-02    1.19e-02
   1  1.423831e+02    3.42e+01    4.07e+02   6.91e-01   1.96e+00  3.00e+04        0    1.36e-02    2.56e-02
   2  1.278684e+02    1.45e+01    2.54e+02   4.03e-01   1.62e+00  9.00e+04        0    1.43e-02    3.98e-02
Robust solving tieration number: 2
iter      cost      cost_change  |gradient|   |step|    tr_ratio  tr_radius  ls_iter  iter_time  total_time
   0  1.002237e+02    0.00e+00    2.43e+02   0.00e+00   0.00e+00  1.00e+04        0    1.10e-02    1.11e-02
   1  9.720633e+01    3.02e+00    1.33e+02   1.54e-01   1.47e+00  3.00e+04        0    1.28e-02    2.39e-02
   2  9.648340e+01    7.23e-01    7.11e+01   8.45e-02   1.41e+00  9.00e+04        0    1.30e-02    3.70e-02
   3  9.633257e+01    1.51e-01    3.78e+01   4.25e-02   1.40e+00  2.70e+05        0    1.26e-02    4.96e-02
   4  9.630302e+01    2.96e-02    2.04e+01   1.50e-02   1.45e+00  8.10e+05        0    1.26e-02    6.22e-02
   5  9.629621e+01    6.81e-03    1.13e+01   4.11e-03   1.52e+00  2.43e+06        0    1.29e-02    7.51e-02
   6  9.629420e+01    2.01e-03    6.56e+00   9.50e-04   1.58e+00  7.29e+06        0    1.29e-02    8.80e-02
   7  9.629349e+01    7.08e-04    3.93e+00   4.28e-04   1.61e+00  2.19e+07        0    1.26e-02    1.01e-01
   8  9.629322e+01    2.74e-04    2.42e+00   4.23e-04   1.63e+00  6.56e+07        0    1.28e-02    1.13e-01
   9  9.629311e+01    1.11e-04    1.52e+00   3.47e-04   1.64e+00  1.97e+08        0    1.27e-02    1.26e-01
---------------------------- motion only ------------------------
Merged summaries:
--------------------------------------------------
Iteration No.0

Solver Summary (v 1.13.0-eigen-(3.3.4)-lapack-suitesparse-(5.1.2)-cxsparse-(3.1.9)-openmp)

                                     Original                  Reduced
Parameter blocks                           64                        1
Parameters                                196                        7
Effective parameters                      195                        6
Residual blocks                            67                       67
Residual                                  132                      132
[... 508 lines cut ...]
[transform_local_cs-12] killing on exit
[transform_local_estimate_prior-11] killing on exit
[transform_local_cs_left_camera_estimate_prior-10] killing on exit
[ INFO] [1593415584.410095520, 114.028298942]: Unloading nodelet /label_dilation_nodelet from manager semantic_labels_nodelet_manager
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[transform_vehicle_left_camera_estimate_prior-9] killing on exit
[transform_local_cs_vehcile_estimate-8] killing on exit
[transform_vehicle_left_camera_estimate-7] killing on exit
[tf2_static_alias_vehicle_to_rig_right_estimate-6] killing on exit
[tf2_static_alias_vehicle_to_rig_left_estimate-5] killing on exit
[tracklets_depth_node-4] killing on exit
[viso_feature_tracking_nodelet_left-3] killing on exit
[ INFO] [1593415584.709337631, 114.028298942]: Unloading nodelet /viso_feature_tracking_nodelet_left from manager feature_matching_nodelet_manager
[gamma_correction_nodelet_left-2] killing on exit
[feature_matching_nodelet_manager-1] killing on exit
--------------------------------------
Dumped map to /tmp/mono_lidar_map_dump.yaml--------------------------------------

shutting down processing monitor...
... shutting down processing monitor complete
done
root@9dca07168881:/workspace/limo_ws/devel# ^C
root@9dca07168881:/workspace/limo_ws/devel# ^C
root@9dca07168881:/workspace/limo_ws/devel#

johannes-graeter, 2020-06-30:
Hm, the ba runs at 3x the speed I would expect from your machine specs...
Perhaps the docker container is the issue (i find it hard to estimate the performance of a virtualized device...) could you try running the install bash script and do the evaluation of 04.bag again?
Thanks for finding this issue and your motivation and time :)

Best,
Johanned

johannes-graeter, 2020-07-26:
Closing due to inactivity.

closed 2020-07-26 by johannes-graeter