johannes-graeter/limo #53: calibration matrix

izzat1, 2020-08-05:
Hi;

I am trying to create my own bag but not sure what to put in tf_static. I am assuming it is the lidar to camera projection but the value in your test bag does not not match what I expect to get from kitti. I am assuming you are using vehicle coordinates.

Thanks
Izzat

johannes-graeter, 2020-08-09:
Hi there,

thanks for the interest in my work :)
I am on vacation at the moment, without computer. I when I am back I will look into it, it dates some time back that I created the bags. On an empty stomach I would assume that the intrinsic and extrinsic calibration are seperate (and not combined in the projection matrix)...

johannes-graeter, 2020-08-09:
I will habe a closer look when I am back at the end of august
Best,
Johannes

izzat1, 2020-08-11:
Thanks for the hint. I think I found out how to do this using Tr matrix from Kitti. I am going to close this. Enjoy your vacation!

closed 2020-08-11 by izzat1

johannes-graeter, 2020-08-11:
Thanks for trying it out :) when you succeed generating the bags from Kitti, it would be great if you could share the code. Due to license reasons I can not publish mine... Best, Johannes

izzat1, 2020-08-13:
Will do. Still not getting the same pose number when I run your bag.

WesG911, 2021-01-08:
Hello @izzat1 ,

Did you succeed generating bags from kitti ? Can you please share more about how you did it ? 

Thank you very much in advance.

Regards,
Wes