johannes-graeter/limo #57: KITTI rosbags

Diggle221, 2020-12-02:
Hi,

Thank you for sharing your work.about LIMO. Can you also share other KITTI rosbags that are appropriate for your code?I have made a few improvements to your project and want to verify the experimental results.

Thank you

johannes-graeter, 2020-12-02:
Hi Diggle,

thanks for the interest in my work! Awesome that you worked to improve this project!!! There is more data, however the bags are quite big and since I am no longer at the University I do not have enough space to host the bigger sequences (f.e. Sequ 0 has around 20 GB for all data)
Actually I would like to also supply a non ros version with a simple main, that reads the data bit by bit but time is sparse for these hobby projects :)

How about you file a pull request and I look over it first?

Thanks again :)
Best,
Johannes

Diggle221, 2020-12-02:
Thank you very much for your reply. I'm sorry for having been disconnected from the school now. I'll pull request to you on GitHub tomorrow morning. Thank you again,thank you.
                                                                Qi Jichao

Diggle221, 2020-12-02:
Thank you very much for your reply!!!!! I'm a postgraduate from Xinjiang University in China. My name is Qi Jichao. I'm not very confident about the effect of my own improvement（For odometry part, I judge the depth of feature points, and adopt different pose calculation methods for
 different feature points）, and I haven't published a paper. Therefore, I'd like to carry out experiments in other kitti rosbags, then send the results of the experiment and code to you. Do you think it's OK?Thanks again!!!!
Best wishes!!
Thanks again!!!!
 Qi Jichao

johannes-graeter, 2020-12-02:
Hi Qi,

awesome, its cool to work together with collegues from all over the world (I myself am living in Germany :) ) 
Ok let me organize sth where i can share the files (also find the files itself ;) on what hard drive did i Save them again ...:)?) . In the mean time you could fork the repo add your changes and do a pull request to my repo so i can review the changes more easily...

Best,
Johannes

johannes-graeter, 2021-01-08:
Sorry Qi, it is really hard for me to find the time to do this. Unfortunately the hard disk with the data on it is corrupt, so if I can not restore it I will need to recalculate the bags. Since I have the impression that most people use LIMO as a whole and not the individual parts, I think the best solution is to make a standalone executable that works without ros and can read the KITTI data in their raw data. If zou are motivated, Some help would be more than welcome :)

Best,
JOhannes

KleinYuan, 2021-11-21:
Do you still plan to implement the interface with the raw KITTI data?
Alternatively, if you can provide the script you use to convert the KITTI raw data to the rosbags, it will be very helpful.

closed 2023-11-09 by johannes-graeter