johannes-graeter/limo #58: Calculating /tf_static and /tf

WesG911, 2021-01-07:
Hi,

Can you please indicate how did you calculate /tf_static and/or /tf used for the provided datasets ?

Thank you very much in advance

johannes-graeter, 2021-01-07:
Hi,

thats the calibration that comes with kitti itself. I only transeferred them from the calib.txt files to tf_static.

Best,
Johannes

WesG911, 2021-01-07:
Thank you very much for your answer.

Can you please clarify for me what is /local_cs ? is it the GPS/IMU frame ?

Best regards,
Wes

johannes-graeter, 2021-01-08:
here we need to distinguish between tf_static which holds the extrinsic calibration as described before and everything that is publsihed via /tf . The /tf topics are all pnon static attributes. Local_cs is  the central coordinate sytem of the car. here /estimate/local_cs is the estimated pose and local_cs is the groundtruth provided by KITTI. This is used for plotting groundtruth next to the estimate in rviz (as hsown in the video in README.md)
Hope that solves your question, reopen if it doesn't.
Best, Johannes

closed 2021-01-08 by johannes-graeter

WesG911, 2021-01-08:
Thank you for your aswer Johannes. 

Actually, I still don't get how were the non static attributes of /tf calculated. 
The provided bag shows that its **frame_id** is **/local_cs** and its **child_frame_id** is **/sensor/camera**.
Can you please clarify more for me how was it calculated from the raw kitti data ? 

Thank you very much in advance, 

Best regards,
Wes