johannes-graeter/limo #59: run error

gp1998-star, 2021-01-08:
Hi ,thank you for sharing your project !
But when i followed your instruction, i got some problems.
`roslaunch demo_keyframe_bundle_adjustment_meta kitti_standalone.launch 
... logging to /home/autolabor/.ros/log/25862922-5189-11eb-947f-8878736befb8/roslaunch-autolabor-host-13251.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Resource not found: matches_conversion_ros_tool
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/autolabor/limo_ws/build_limo_release/catkin_tools_prebuild
ROS path [2]=/home/autolabor/slam_ws/src
ROS path [3]=/opt/ros/melodic/share
The traceback for the exception was written to the log file
`

gp1998-star, 2021-01-08:
I successfully compiled limo, and I encountered this error when I run kitty04.bag.
i have installed all the requirements include ceres before build.
i'll appreciate if you can reply !

johannes-graeter, 2021-01-08:
matches_conversion_ros_tool is one of the execs I wrote... Seems as though it is not downloaded in your case, I need to investigate this. In the mean time you could try building the docker image and tell me if that did work...

Best, Johannes

gp1998-star, 2021-01-08:
I found it in the build. Does it mean it was installed correctly?
[image]

gp1998-star, 2021-01-08:
> matchs_conversion_ros_tool是我写的执行程序之一...好像没有下载，您需要对此进行调查。同时，您可以尝试构建docker映像，并告诉我它是否有效...
> 
> 最好，约翰内斯

I really want to use docker, but my pip install pykitti \ jupyter \ jupytext --upgrade cannot be executed smoothly due to network reasons (china), can you tell me how to replace pip install?

gp1998-star, 2021-01-08:
> matchs_conversion_ros_tool

When I compile, matchs_conversion_ros_tool may be ignored, but I found it in the build_limo_release folder, and it is not in src. If I have not installed it, can you tell me how to install this executable?

johannes-graeter, 2021-01-08:
Did you source the catkin workspace as described in the readme? Ypu need to add that to your bashrc (or do it each time manually :))

gp1998-star, 2021-01-09:
> Did you source the catkin workspace as described in the readme? Ypu need to add that to your bashrc (or do it each time manually :))

I solved this problem, I should use setup. bash instead of setup. sh.  thanks a lot !!!

closed 2021-01-09 by gp1998-star