johannes-graeter/limo #62: Bulid Problem

yanlong658, 2021-07-30:
Hello

Your work is very nice.
However, I have some question about the build.
I do my best to deal with it but I don't any ideal.
Could you help me ?
[image]

Thank you

LongruiDong, 2021-11-08:
> Hello
> 
> Your work is very nice. However, I have some question about the build. I do my best to deal with it but I don't any ideal. Could you help me ? [image]
> 
> Thank you

i am the same with you. Have u solved yet?

LongruiDong, 2021-11-09:
> Hello
> 
> Your work is very nice. However, I have some question about the build. I do my best to deal with it but I don't any ideal. Could you help me ? [image]
> 
> Thank you

here is an answer from author:

https://github.com/johannes-graeter/limo/issues/63#issuecomment-963084494

giewogun, 2022-07-31:
> > Hello
> > Your work is very nice. However, I have some question about the build. I do my best to deal with it but I don't any ideal. Could you help me ? [image]
> > Thank you
> 
> here is an answer from author:
> 
> [#63 (comment)](https://github.com/johannes-graeter/limo/issues/63#issuecomment-963084494)

Sorry, could u be more specific where can I find the jupyter related stuff that I should comment out?  THX!

johannes-graeter, 2022-07-31:
in https://github.com/johannes-graeter/limo/blob/master/docker/src/Dockerfile lines 84 and 85 also consider the internet connection related issues that are mentioned more below in the linked issue.

closed 2023-11-09 by johannes-graeter