johannes-graeter/limo #63: build error in docker

LongruiDong, 2021-11-08:
Hi,
Thanks for presenting such a great algorithm.

I follow the guide to build in docker, but met error:

```
(py27) :~/Project/limo/docker$ sudo docker-compose build limo
Building limo
Step 1/15 : FROM ros:melodic-perception
 ---> 0eb9859ebf79
Step 2/15 : RUN apt update && apt install --no-install-recommends -y     build-essential     cmake     libopencv-dev     libgoogle-glog-dev     libatlas-base-dev     libeigen3-dev     libsuitesparse-dev     libpng++-dev     git     libx11-6     libgtk2.0-0     vim     nano     wget     python-catkin-tools     ros-melodic-opencv-apps     ros-melodic-diagnostic-updater     ros-melodic-tf-conversions     lcov     tmux     python-pip     && apt clean     && rm -rf /var/lib/apt/lists/*     && rm -rf /tmp/*
 ---> Using cache
 ---> 969accae78e4
Step 3/15 : RUN pip install pykitti     jupyter     jupytext --upgrade
 ---> Running in 253a66d70fb1
Collecting pykitti
  Downloading https://files.pythonhosted.org/packages/c3/7e/e2b71b9221ccb8503dce9673e3cb36edcdc314ec374f9e69e6d38ed1abde/pykitti-0.3.1-py2.py3-none-any.whl
Collecting jupyter
  Downloading https://files.pythonhosted.org/packages/83/df/0f5dd132200728a86190397e1ea87cd76244e42d39ec5e88efd25b2abd7e/jupyter-1.0.0-py2.py3-none-any.whl
Collecting jupytext
  Downloading https://files.pythonhosted.org/packages/73/3c/e98e8b105f0d389bb166886878e18e43472cd324a1b8036a589077781f78/jupytext-1.8.1.tar.gz (659kB)
Collecting numpy (from pykitti)
  Downloading https://files.pythonhosted.org/packages/3a/5f/47e578b3ae79e2624e205445ab77a1848acdaa2929a00eeef6b16eaaeb20/numpy-1.16.6-cp27-cp27mu-manylinux1_x86_64.whl (17.0MB)
Collecting pandas (from pykitti)
  Downloading https://files.pythonhosted.org/packages/db/83/7d4008ffc2988066ff37f6a0bb6d7b60822367dcb36ba5e39aa7801fda54/pandas-0.24.2-cp27-cp27mu-manylinux1_x86_64.whl (10.1MB)
Collecting Pillow (from pykitti)
  Downloading https://files.pythonhosted.org/packages/12/ad/61f8dfba88c4e56196bf6d056cdbba64dc9c5dfdfbc97d02e6472feed913/Pillow-6.2.2-cp27-cp27mu-manylinux1_x86_64.whl (2.1MB)
Collecting matplotlib (from pykitti)
  Downloading https://files.pythonhosted.org/packages/9d/40/5ba7d4a3f80d39d409f21899972596bf62c8606f1406a825029649eaa439/matplotlib-2.2.5-cp27-cp27mu-manylinux1_x86_64.whl (12.8MB)
Collecting nbconvert (from jupyter)
  Downloading https://files.pythonhosted.org/packages/79/6c/05a569e9f703d18aacb89b7ad6075b404e8a4afde2c26b73ca77bb644b14/nbconvert-5.6.1-py2.py3-none-any.whl (455kB)
Collecting ipywidgets (from jupyter)
  Downloading https://files.pythonhosted.org/packages/6b/bb/285066ddd710779cb69f03d42fa72fbfe4352b4895eb6abab551eae1535a/ipywidgets-7.6.5-py2.py3-none-any.whl (121kB)
Collecting qtconsole (from jupyter)
  Downloading https://files.pythonhosted.org/packages/fb/ef/6cf8a7301726a7b7149e95102c70f809542dfcd916826e8dccc0d90b2605/qtconsole-5.0.0.tar.gz (424kB)
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-mHOTWP/qtconsole/setup.py", line 18
        print(error, file=sys.stderr)
                         ^
    SyntaxError: invalid syntax
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-mHOTWP/qtconsole/
ERROR: Service 'limo' failed to build: The command '/bin/sh -c pip install pykitti     jupyter     jupytext --upgrade' returned a non-zero code: 1

```

seems like  python version? but how to get over it?
Thanks 😁

johannes-graeter, 2021-11-08:
Hi a first guess would be to just comment out jupyter related stuff, this seems to be for the ui interfaces.
Sorry I can't test it myself, no time at all ;)

Best
Johannes

LongruiDong, 2021-11-09:
> Hi a first guess would be to just comment out jupyter related stuff, this seems to be for the ui interfaces. Sorry I can't test it myself, no time at all ;)
> 
> Best Johannes

Yeah, I comment jupter related package and build over this line.
however, always encounter git clone error:

```
Cloning into 'limo'...
fatal: unable to access 'https://github.com/johannes-graeter/limo.git/': gnutls_handshake() failed: The TLS connection was non-properly terminated.

```

even I have  add
```
RUN unset https_proxy \
&& unset http_proxy

RUN git config --global http.postBuffer 1048576000
```
,still  not work

LongruiDong, 2022-03-22:
> thx a lot for you contrib @LongruiDong in #66 . did this fix this issue as well?

my previous problem are due to internet reasons(CHINA mainland), I split git clone to individual commands, I have tried several times via
```
# use cache when building the image
docker-compose build limo
```

finally git clone all succeed

johannes-graeter, 2022-03-22:
nice!

closed 2022-03-22 by johannes-graeter