johannes-graeter/limo #67: can not use rviz in docker

LongruiDong, 2022-03-22:
Hi~

after build the docker, when I setp to
```
rviz -d ${your_catkin_workspace}/src/demo_keyframe_bundle_adjustment_meta/res/default.rviz

```

returns:

```
rviz -d src/limo/demo_keyframe_bundle_adjust│
ment_meta/res/default.rviz                                                        │setting /run_id to c4eb752c-a9b1-11ec-b32b-0242ac120004
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'        │process[rosout-1]: started with pid [975]
qt.qpa.screen: QXcbConnection: Could not connect to display                       │started core service [/rosout]
Could not connect to any X display.
```

How could view rviz window with docker?

johannes-graeter, 2022-03-22:
oh yeah thats a tricky one. You need to share the X server of the docker container as described here https://medium.com/intro-to-artificial-intelligence/rviz-on-docker-bdf4d0fca5b

LongruiDong, 2022-03-22:
> 

I am quite new to docker, is there simpler way by modifing docker-compose only?

after reading that blog and [this](https://www.cloudsavvyit.com/10520/how-to-run-gui-applications-in-a-docker-container/) , what I understand:

```
version: '3.4'

services:
  limo:
    container_name: limo
    build:
      context: ./src
    image: ros:melodic-perception
    ports:
      - 8888:8888
    environment:
      - DISPLAY=${DISPLAY}
    volumes:
      - $HOME/Project/limo/limo_data:/limo_data 
      - /tmp/.X11-unix:/tmp/.X11-unix
    network_mode: host
    stdin_open: true
    tty: true
```

LongruiDong, 2022-03-22:
> > 
> 
> I am quite new to docker, is there simpler way by modifing docker-compose only?
> 
> after reading that blog and [this](https://www.cloudsavvyit.com/10520/how-to-run-gui-applications-in-a-docker-container/) , what I understand: below is docker-compose.yml
> 
> ```
> version: '3.4'
> 
> services:
>   limo:
>     container_name: limo
>     build:
>       context: ./src
>     image: ros:melodic-perception
>     ports:
>       - 8888:8888
>     environment:
>       - DISPLAY=${DISPLAY}
>     volumes:
>       - $HOME/Project/limo/limo_data:/limo_data 
>       - /tmp/.X11-unix:/tmp/.X11-unix
>     network_mode: host
>     stdin_open: true
>     tty: true
> ```

lucky!

I change docker-compose.yml like this:
```
version: '3.4'

services:
  limo:
    container_name: limo
    build:
      context: ./src
    image: ros:melodic-perception
    ports:
      - 8888:8888
    environment:
      - DISPLAY=${DISPLAY}
    volumes:
      - $HOME/limo_data:/limo_data 
      - /tmp/.X11-unix:/tmp/.X11-unix
    # network_mode: host #  error: conflicting options: host type networking can't be used with links comment to avoid it
    stdin_open: true
    tty: true

```
rebuild and I can invoke rviz normally

johannes-graeter, 2022-03-22:
awesome! can you PR this plz?
THanks for you time invest!

closed 2022-03-22 by LongruiDong