johannes-graeter/limo #70: Hackaton

vedin-klovo, 2022-06-05:
Hi Johannes,

I implemented this algorithm for Carla a few years and would like to help with future development. Couldn't find you email so I created this issue. You can email me at vedin13@gmail.com (personal) or vklovo1@etf.unsa.ba (university).

Best regards,
Vedin

johannes-graeter, 2022-06-05:
Hi vedin,

this is awesome!!! Thanks for being the first to reply to my call for hackathon :).
Lets do this! Would love to just sit down for a couple of hours and do some paired coding, corresponding to the tasks i mentioned in the readme.
For the timing however i will need to push the date back a little. I am at the moment renovating my apartment and then on holidays and moving in. So how about we start in july? If your fingers tickle you can also start by yourself. Whats your skill set and field of expertise?

Thanks and all the best johannes

vedin-klovo, 2022-06-06:
Hi Johannes,

Sounds great, I'm looking forward to getting started. I can start now as well, but I would need some guidance to get started. If things are busy at your end, it's no problem to wait until July. If there's already a document/plan which I can use to start myself, I could do start earlier as well.

By education, I'm finishing my final years of Master's studies at the Faculty of Electrical Engineering in Sarajevo, Bosnia, at the Department of Control and Electronics. Personally, I learn and work in Robotics Vision and Perception, and I'm looking to sharpen my skills in the fields of VO, SLAM, 3D reconstruction, and mapping. I did some work in ARCore as well. It's kind of tough to get into this field all alone, so I'm looking forward to learning from an expert!

From the technical side of things, I'm proficient in Python, C++, and Java. I'm also quite good at embedded programming. 

Best regards,
Vedin

vedin-klovo, 2022-06-06:
> Awesome!
> *
> When you do some embedded you problem know your way around plain c as well? I actually wanted to have a plain c interface for the library for a long time now in order to interface with python or rust. I tried to do it with boost python and c++ but found that rather annoying.
> *
> When i did this library i was purely working in ros, but it turned out to be a mistake to only interface through ros with f e. Kitti. It could be a good first task just create an executable form the library that runs it on kitti without ros.
> Gladly I would have a meeting of an hour with you or so to discuss that and a working plan.
> How does that sound?
> Best

1. In general, I'm familiar with C, it was my first programming language, actually, but somehow most of my embedded work was C++. I think I could manage this, though.
2. Sounds good

I'm fairly available this week. Does 12:00 CET Tuesday, 10:30 CET Wednesday (or any time after that) or any morning/around noon time on Thursday work for you? Looking forward to speaking! 

Best regards,
Vedin

YoshuaNava, 2022-06-13:
Hi,

I think the work from this repo is excellent. I unfortunately can't engage in open source projects at the moment, but wish you the best in this sprint.

Something I would suggest based on personal experience is to split ROS and core algorithms, but not in a way in which observability is completely lost, e.g. removing publishers and logging from places where they help to retrieve useful feedback for algorithm development.

I would propose a candidate design pattern for implementing ROS nodes: https://refactoring.guru/design-patterns/template-method

With this, you could have a parent class Limo, and then children classes LimoRos, LimoRos2, etc, which implement all the infrastructure to visualize data as private methods. Also, you could preserve the threading model that you need for Limo, while enabling observability in the interface.

johannes-graeter, 2022-06-20:
This is great advice @YoshuaNava thx, will definitely consider that!
@vedin-klovo sorry for not replying, how about we have a call on thursday (1pm would be cool so that its in my lunchbreak)