johannes-graeter/limo #71: can not play rosbag in docker

giewogun, 2022-07-31:
Hi Johannes, thank you for sharing your project !

I met a 'newbie' problem when I tried to use docker to run the project, I've run the docker and **1. roscore** then **2. play the rosbag**, but the terminal said **Failed to contact master at [localhost:11311].  Retrying...**. It seems the two terminals are not connected with each other? 

Moreover, when I **roslaunch**, the terminal said:

[ INFO] [1659300135.786372846]: Parameter '/mono_lidar/diag_pub_topic' is not defined. Setting default value.
[ INFO] [1659300135.788008240]: Parameter '/mono_lidar/diagnostic_updater_name' is not defined. Setting default value.
[ INFO] [1659300135.788980394]: Parameter '/mono_lidar/diagnostic_updater_hardware_id' is not defined. Setting default value.
[ INFO] [1659300135.789820873]: Parameter '/mono_lidar/diagnostic_updater_rate' is not defined. Setting default value.
[ INFO] [1659300135.790664991]: Parameter '/mono_lidar/diagnostic_updater_rate_tolerance' is not defined. Setting default value.

Did anyone meet the same issue before? Thanks in advance!

Toni

johannes-graeter, 2022-07-31:
Hi there,

after you drop in the docker container, do you call tmux? Since the container is not exposed to your local system, all terminals must be called in the same container, the simplest is to call tmux in it and then open up new terminals and switch between them

giewogun, 2022-07-31:
Hi Johannes,
thx for ur reply, now I can use the tmux to switch between terminals, but in the terminal when I **cd into /limo_data folder**, und **run rosbag play** I got the error below, it seems the rosbag in my local system is not linked to the docker?

root@b159446c6461:/limo_data# rosbag play 01.bag -r 0.1 --pause --clock            
[ INFO] [1659305708.679770224]: Opening 01.bag                                     
[FATAL] [1659305708.691991975]: Error opening file: 01.bag     

Thx!

johannes-graeter, 2022-08-01:
yes, you need to say docker which folder is visible inside the container. Per default this is $HOME/limo_data (as can be seen in the docker-compose.yml). If you put it there it should be visible inside the container.

giewogun, 2022-08-01:
> 

what I think is if I put the bag inside this limo_data folder in $HOME, then with this yml file, I can see the bag files in docker through "cd /limo_data" then "ls" command, right?
But for now its not the case, kinda weird.

closed 2023-11-09 by johannes-graeter