johannes-graeter/limo #73: Visualize odometry data like the youtube video

Abdob, 2022-08-11:
How can we visualize the odometry data like the youtube video:

[image]

I have tried setting path and odometry topics but didn't get the paths to display

Is my output below expected?

[image]

shb512156211, 2024-03-18:
Have you deal with the problem?@Abdob

johannes-graeter, 2024-03-18:
there are default.rviz files which should cover that. However at the moment the project is undergoing a general refurbish, so things might temproraly not work as expected...