johannes-graeter/limo #74: Reproducing results for sequence 1

Abdob, 2022-08-11:
Using the bag file for sequence one, getting the pose result and plotting on matlab I get this:
[image]

our results the rover traveled around 100+meters but the plot shows otherwise:

[image]

Abdob, 2022-08-13:
I'm trying a more powerful PC and proper plotting, this is what I get:
[image]

Abdob, 2022-08-13:
When increasing the rate to 0.125 12 samples go missing, in both cases the odometry becomes dense towards the end
[image]

At around sample 200/1100 X and Z level off, this hiccup can be seen when plotting sample number vs X or Z

Abdob, 2022-08-13:
I get some improvements when configuring the odometry application by uncommenting the catkin config related lines in the dockerfile
[image]
[image]

Now, at near sample 500 the XZ trajectory is leveling off..

johannes-graeter, 2023-11-09:
not reproducable so closing

closed 2023-11-09 by johannes-graeter