johannes-graeter/limo #75: build error with ros melodic

Huyhust13, 2022-08-16:
Dear,
I faced some error when compile this project in ros melodic. It cause nodelet different from kinetic to melodic.
I found out solution from this link: https://github.com/KumarRobotics/msckf_vio/issues/69
Finally, it works!

johannes-graeter, 2022-09-05:
awesome! would be great if you could add this change and file a pull request to contribute. In order to maintain compatibilty with kinetic you would need to detect if its a kinetic or melodic build though...

Best Johannes

Huyhust13, 2022-09-05:
> awesome! would be great if you could add this change and file a pull request to contribute. In order to maintain compatibilty with kinetic you would need to detect if its a kinetic or melodic build though...
> 
> Best Johannes

There is already pull request: https://github.com/KIT-MRT/feature_tracking/pull/1/commits/8066fdce1b585bc5d25e25baed58a8f71b3d6433

johannes-graeter, 2022-09-05:
Am I correct

Huyhust13, 2022-09-05:
I didn't test it in kinetic :D

closed 2024-03-18 by johannes-graeter