johannes-graeter/limo #79: Hi,I meet with this problem by using"docker-compose build"

rodrickcui, 2022-09-30:
Unsupported config option for services.limo: 'runtime'
How to solve it?
the content of docker-compose.yml is 
version: '3.4'

services:
  limo:
    container_name: limo
    build:
      context: ./src
    image: ros:melodic-perception
    ports:
      - 8888:8888
    network_mode: host
    ipc: host
    privileged: true  
    runtime: nvidia

rodrickcui, 2022-10-05:
solved

y-l-a, 2023-08-09:
Hi! I have also encountered this problem. How did you solve it?

johannes-graeter, 2023-08-09:
did you install the nvidia-runtime? https://stackoverflow.com/questions/69995187/docker-compose-unsupported-config-option-for-runtime
 NOt sure if the image the container is based on (ros:melodic-perception) now uses gpu support...

closed 2023-11-09 by johannes-graeter