johannes-graeter/limo #81: roslaunch question

xiang-1208, 2023-01-21:
When I was running with roslaunch, the node had an error and shut down after I played the 04.bag in the example. Why?
[image]

johannes-graeter, 2023-01-23:
can you share the logfile?

xiang-1208, 2023-01-23:
[roslaunch-xiang-3200.log](https://github.com/johannes-graeter/limo/files/10478261/roslaunch-xiang-3200.log)

johannes-graeter, 2023-01-23:
Thanks for the uqick repsonse. this is not the correct one, I meant the one that is mentioned in the error message you posted.

xiang-1208, 2023-01-25:
[5be05846-9c9e-11ed-8b38-0242866b5d7c.zip](https://github.com/johannes-graeter/limo/files/10498624/5be05846-9c9e-11ed-8b38-0242866b5d7c.zip)
I don't find the correct one, I send you the all logfile from this run, thank you for you patient help.
`... logging to /home/xiang/.ros/log/5be05846-9c9e-11ed-8b38-0242866b5d7c/roslaunch-xiang-15796.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://xiang:36775/

SUMMARY
========

PARAMETERS
 * /add_outlier_flag_node/has_depth: True
 * /add_outlier_flag_node/publisher_topic: /tracklets_depth/...
 * /add_outlier_flag_node/subscriber_depth_topic: /tracklets_depth/...
 * /add_outlier_flag_node/verbosity: info
 * /gamma_correction_nodelet_left/auto_gamma: True
 * /gamma_correction_nodelet_left/gamma: 1.2
 * /label_dilation_nodelet/half_kernel_size: 8
 * /label_dilation_nodelet/publisher_topic: /semantic_labels/...
 * /label_dilation_nodelet/subscriber_topic: /sensor/camera/co...
 * /label_dilation_nodelet/verbosity: info
 * /mono_lidar/active_path_publisher_topic: /estimate/active_...
 * /mono_lidar/calib_source_frame_id: estimate/vehicle
 * /mono_lidar/calib_target_frame_id: estimate/sensor/c...
 * /mono_lidar/camera_info_subscriber_topic: /sensor/camera/gr...
 * /mono_lidar/critical_rotation_difference: 0.03
 * /mono_lidar/dump_path: /tmp/poses_dump.txt
 * /mono_lidar/height_over_ground: 0.31
 * /mono_lidar/landmarks_publisher_topic: /estimate/landmarks
 * /mono_lidar/max_number_landmarks_far_bin: 100
 * /mono_lidar/max_number_landmarks_middle_bin: 200
 * /mono_lidar/max_number_landmarks_near_bin: 200
 * /mono_lidar/max_size_optimization_window: 12
 * /mono_lidar/max_solver_time: 0.15
 * /mono_lidar/min_median_flow: 5.0
 * /mono_lidar/min_number_connecting_landmarks: 18
 * /mono_lidar/outlier_labels_yaml: /home/xiang/Proje...
 * /mono_lidar/outlier_rejection_num_iterations: 1
 * /mono_lidar/outlier_rejection_quantile: 0.95
 * /mono_lidar/path_publisher_topic: /estimate/complet...
 * /mono_lidar/prior_global_frame: estimate_prior/lo...
 * /mono_lidar/prior_speed: 5.0
 * /mono_lidar/prior_vehicle_frame: 
 * /mono_lidar/robust_loss_depth_thres: 0.16
 * /mono_lidar/robust_loss_reprojection_thres: 1.6
 * /mono_lidar/show_debug_image: False
 * /mono_lidar/shrubbery_weight: 0.9
 * /mono_lidar/tf_child_frame_id: estimate/vehicle
 * /mono_lidar/tf_parent_frame_id: estimate/local_cs...
 * /mono_lidar/tf_waiting_time: 0.1
 * /mono_lidar/time_between_keyframes_sec: 0.4
 * /mono_lidar/tracklets_subscriber_topic: /tracklets_depth/...
 * /mono_lidar/trajectory_publisher_topic: /estimate/trajectory
 * /mono_lidar/verbosity: info
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /semantic_labels_nodelet/matches_publisher_topic: /tracklets_depth/...
 * /semantic_labels_nodelet/roi_height: 3
 * /semantic_labels_nodelet/roi_width: 3
 * /semantic_labels_nodelet/subscriber_labels_topic: /semantic_labels/...
 * /semantic_labels_nodelet/subscriber_matches_topic: /tracklets_depth/...
 * /semantic_labels_nodelet/verbosity: debug
 * /tf2_static_alias_vehicle_to_rig_left_estimate/from_source_frame_id: sensor/camera/gra...
 * /tf2_static_alias_vehicle_to_rig_left_estimate/from_target_frame_id: sensor/camera
 * /tf2_static_alias_vehicle_to_rig_left_estimate/to_source_frame_id: estimate/sensor/c...
 * /tf2_static_alias_vehicle_to_rig_left_estimate/to_target_frame_id: estimate/sensor/c...
 * /tf2_static_alias_vehicle_to_rig_right_estimate/from_source_frame_id: sensor/camera/gra...
 * /tf2_static_alias_vehicle_to_rig_right_estimate/from_target_frame_id: sensor/camera
[... 278 lines cut ...]
[semantic_labels_nodelet-18] killing on exit
[label_dilation_nodelet-17] killing on exit
[semantic_labels_nodelet_manager-16] killing on exit
[add_outlier_flag_node-15] killing on exit
[ INFO] [1674644306.355951243, 11.091428711]: Unloading nodelet /semantic_labels_nodelet from manager semantic_labels_nodelet_manager
[transform_local_cs_camera_gt-14] killing on exit
[transform_vehicle_camera_vehicle_gt-13] killing on exit
[transform_local_cs-12] killing on exit
[transform_local_estimate_prior-11] killing on exit
[transform_local_cs_left_camera_estimate_prior-10] killing on exit
[ INFO] [1674644306.356727961, 11.091428711]: Unloading nodelet /label_dilation_nodelet from manager semantic_labels_nodelet_manager
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[transform_vehicle_left_camera_estimate_prior-9] killing on exit
[transform_local_cs_vehcile_estimate-8] killing on exit
[transform_vehicle_left_camera_estimate-7] killing on exit
[tf2_static_alias_vehicle_to_rig_right_estimate-6] killing on exit
[tf2_static_alias_vehicle_to_rig_left_estimate-5] killing on exit
[tracklets_depth_node-4] killing on exit
--------------------------------------
Dumped map to /tmp/mono_lidar_map_dump.yaml--------------------------------------

shutting down processing monitor...
... shutting down processing monitor complete
done`

MinaMedhat12, 2023-11-09:
Hello,

Did you find a solution to this problem as I have the same one now?

johannes-graeter, 2023-11-09:
can you share a log dump?

MinaMedhat12, 2023-11-09:
Thank you for the quick response.

I have tried it on Ubuntu 20.04 and it's working perfectly now.

Just a question in Rviz the lidar topics always give me an error and it doesn't show up
"Message 
Failed to transform from frame [/sensor/velodyne] to frame [estimate/local_cs_vehicle]" 
but it does the trajectory correct 
https://drive.google.com/file/d/1NVM4j72VZs1yKqm9mqDe0tOSttir7QBi/view?usp=sharing