johannes-graeter/limo #83: docker build error

where2go947, 2023-12-20:
hello, thanks for your awesome work!
I'm trying to use docker to build on Ubuntu 20.04, Docker version 20.10.21, but errors occurred:
```
$ docker-compose build limo
[+] Building 124.3s (16/20)                                      docker:default
 => [limo internal] load build definition from Dockerfile                  0.0s
 => => transferring dockerfile: 3.92kB                                     0.0s
 => [limo internal] load .dockerignore                                     0.0s
 => => transferring context: 2B                                            0.0s
 => [limo internal] load metadata for docker.io/library/ros:melodic-perce  1.5s
 => [limo internal] load build context                                     0.0s
 => => transferring context: 29B                                           0.0s
 => [limo  1/16] FROM docker.io/library/ros:melodic-perception@sha256:a2f  0.0s
 => CACHED [limo  2/16] RUN apt update && apt install --no-install-recomm  0.0s
 => CACHED [limo  3/16] RUN pip install --upgrade pip                      0.0s
 => [limo  4/16] RUN pip install pykitti                                  53.2s
 => [limo  5/16] RUN apt update && apt install --no-install-recommends -  40.2s
 => [limo  6/16] RUN rm -rf /workspace     && mkdir -p /workspace/limo_ws  0.5s 
 => [limo  7/16] WORKDIR /workspace/limo_ws/src                            0.0s 
 => [limo  8/16] RUN rm -rf /workspace/limo_ws/src/limo     && git clone  23.4s 
 => [limo  9/16] RUN cp -rf /usr/include/eigen3/Eigen /usr/include/Eigen   0.4s 
 => [limo 10/16] RUN ln -sf /usr/include/eigen3/Eigen /usr/include/Eigen   0.5s 
 => [limo 11/16] WORKDIR /workspace/limo_ws/                               0.0s 
 => ERROR [limo 12/16] RUN /bin/bash -c 'source /opt/ros/melodic/setup.ba  4.6s 
------                                                                          
 > [limo 12/16] RUN /bin/bash -c 'source /opt/ros/melodic/setup.bash && catkin build':                                                                          
0.567 NOTICE: Could not determine the width of the terminal. A default width of 80 will be used. This warning will only be printed once.
0.750 -------------------------------------------------------
0.750 Profile:                     default
0.750 Extending:             [env] /opt/ros/melodic
0.750 Workspace:                   /workspace/limo_ws
0.750 -------------------------------------------------------
0.750 Build Space:        [exists] /workspace/limo_ws/build
0.750 Devel Space:        [exists] /workspace/limo_ws/devel
0.750 Install Space:      [unused] /workspace/limo_ws/install
0.750 Log Space:         [missing] /workspace/limo_ws/logs
0.750 Source Space:       [exists] /workspace/limo_ws/src
0.750 DESTDIR:            [unused] None
0.750 -------------------------------------------------------
0.750 Devel Space Layout:          linked
0.750 Install Space Layout:        None
0.750 -------------------------------------------------------
0.750 Additional CMake Args:       None
0.750 Additional Make Args:        None
0.750 Additional catkin Make Args: None
0.750 Internal Make Job Server:    True
0.750 Cache Job Environments:      False
0.750 -------------------------------------------------------
0.750 Whitelisted Packages:        None
0.750 Blacklisted Packages:        None
0.750 -------------------------------------------------------
0.750 Workspace configuration appears valid.
0.750 
0.750 NOTE: Forcing CMake to run for each package.
0.750 -------------------------------------------------------
0.750 [build] Found '23' packages in 0.0 seconds.                                    
0.750 [build] Updating package table.                                                
0.750 Starting >>> catkin_tools_prebuild                                             
0.750 [build 0.2 s] [0/21 complete] [1/8 jobs] [0 queued] [catkin_tools_prebuildFinished <<< catkin_tools_prebuild                               [ 1.8 seconds ]
2.454 Starting >>> mrt_cmake_modules                                                 
2.454 Starting >>> rosinterface_handler                                              
2.454 Starting >>> tracklets_depth                                                   
2.454 [build 1.9 s] [1/21 complete] [3/8 jobs] [2 queued] [mrt_cmake_modules:loa_______________________________________________________________________________
3.197 Errors << tracklets_depth:cmake /workspace/limo_ws/logs/tracklets_depth/build.cmake.000.log
3.197 CMake Error at /workspace/limo_ws/src/mono_lidar_depth/tracklets_depth/CMakeLists.txt:9 (find_package):
3.197   By not providing "Findmrt_cmake_modules.cmake" in CMAKE_MODULE_PATH this
3.197   project has asked CMake to find a package configuration file provided by
3.197   "mrt_cmake_modules", but CMake did not find one.
3.197 
3.197   Could not find a package configuration file provided by "mrt_cmake_modules"
3.197   with any of the following names:
3.197 
3.197     mrt_cmake_modulesConfig.cmake
3.197     mrt_cmake_modules-config.cmake
3.197 
3.197   Add the installation prefix of "mrt_cmake_modules" to CMAKE_PREFIX_PATH or
3.197   set "mrt_cmake_modules_DIR" to a directory containing one of the above
3.197   files.  If "mrt_cmake_modules" provides a separate development package or
3.197   SDK, be sure it has been installed.
3.197 
3.197 
3.197 cd /workspace/limo_ws/build/tracklets_depth; catkin build --get-env tracklets_depth | catkin env -si  /usr/bin/cmake /workspace/limo_ws/src/mono_lidar_depth/tracklets_depth --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/workspace/limo_ws/devel/.private/tracklets_depth -DCMAKE_INSTALL_PREFIX=/workspace/limo_ws/install; cd -
...............................................................................
3.197 Failed << tracklets_depth:cmake                                [ Exited with code 1 ]
3.257 Failed <<< tracklets_depth                                     [ 0.7 seconds ] 
3.257 Abandoned <<< image_preproc                                       [ Unrelated job failed ]
3.257 Abandoned <<< matches_msg_depth_ros                               [ Unrelated job failed ]
3.257 Abandoned <<< matches_msg_ros                                     [ Unrelated job failed ]
3.257 Abandoned <<< matches_msg_types                                   [ Unrelated job failed ]
3.257 Abandoned <<< matches_msg_conversions_ros                         [ Unrelated job failed ]
3.257 Abandoned <<< monolidar_fusion                                    [ Unrelated job failed ]
3.257 Abandoned <<< robust_optimization                                 [ Unrelated job failed ]
3.257 Abandoned <<< keyframe_bundle_adjustment                          [ Unrelated job failed ]
3.257 Abandoned <<< image_preproc_ros_tool                              [ Unrelated job failed ]
3.257 Abandoned <<< matches_conversion_ros_tool                         [ Unrelated job failed ]
3.257 Abandoned <<< util_nodes_tf2_ros_tool                             [ Unrelated job failed ]
3.257 Abandoned <<< viso2                                               [ Unrelated job failed ]
3.257 Abandoned <<< feature_tracking_core                               [ Unrelated job failed ]
3.257 Abandoned <<< tracklets_depth_ros_tool                            [ Unrelated job failed ]
3.257 Abandoned <<< viso_feature_tracking_ros_tool                      [ Unrelated job failed ]
3.257 Abandoned <<< keyframe_bundle_adjustment_ros_tool                 [ Unrelated job failed ]
3.257 Abandoned <<< demo_keyframe_bundle_adjustment_meta                [ Unrelated job failed ]
3.257 [build 2.7 s] [2/21 complete] [2/8 jobs] [0 queued] [1 failed] [rosinterfaFinished <<< mrt_cmake_modules                                   [ 1.4 seconds ]
3.858 [build 3.3 s] [3/21 complete] [1/8 jobs] [0 queued] [1 failed] [rosinterfaFinished <<< rosinterface_handler                                [ 2.0 seconds ]
4.515 [build] Summary: 3 of 21 packages succeeded.                                   
4.515 [build] Ignored: 3 packages were skipped or are blacklisted.                   
4.515 [build] Warnings: None.                                                        
4.515 [build] Abandoned: 17 packages were abandoned.                                 
4.515 [build] Failed: 1 packages failed.                                             
4.515 [build] Runtime: 3.9 seconds total.                                            
4.515 [build] Note: Workspace packages have changed, please re-source setup files to use them.
------
failed to solve: process "/bin/sh -c /bin/bash -c 'source /opt/ros/melodic/setup.bash && catkin build'" did not complete successfully: exit code: 1

```

It seems to be with "mrt_cmake_modules", I run `docker-compose build limo` in the path `limo/docker` 
Could you please give some help? thanks!

johannes-graeter, 2023-12-21:
Tha ks for the interest in my project. At the moment I am doing a major refactor of the project since many dependencies changed to the point where the docker failed to build. Hope to be able to be ready in the first week of january.

Merry Christmas
Johannes

where2go947, 2023-12-24:
@johannes-graeter Thanks!
Merry Christmas

sanersbug, 2024-01-12:
@johannes-graeter I meet the same error, could you please give some advice, thanks